hzeller/rpi-rgb-led-matrix #251: flicker led-matrix

ffqqtv, 2016-12-02:
don't use command line --led-no-hardware-pulse , but it still flickers as : 
https://www.youtube.com/watch?v=aVOCVwvdYg0&feature=youtu.be
Thanks for watching !

hzeller, 2016-12-02:
The glitches you see on the middle left look more like too fast clocking. Make sure to provide a solid power supply and play with higher values of `--led-slowdown-gpio`

Are you using the active3 hardware adapter ?

ffqqtv, 2016-12-05:
Thanks for replying   ! i dit it . Now , i need to display the video using camera (using live)  . With available hardware (adapter of you ) and software(library of you ) , Can they do that ?

hzeller, 2016-12-05:
If you get pixels out of your video stream with libraries you prefer, you can simply fill the Canvas with the pixels you get in a loop.

ffqqtv, 2016-12-08:
Thanks for replying ! i need to send video by VLC and display on RGB matrix .
Thanks for watching !

hzeller, 2016-12-08:
That should be fairly straight-forward. For that, I'd actually use the FlaschenTaschen protocol, as the latest VLC provides that as output, and I have build a FlaschenTaschen implementation that works with the rpi-rgb-led-matrix.

```
$ vlc --vout flaschen --flaschen-display=<hostname> \
      --flaschen-width=160 --flaschen-height=96 \
      <video-filename-or-YouTube-URL>
```

https://github.com/hzeller/flaschen-taschen/tree/master/server#rgb-matrix-panel-display

ffqqtv, 2016-12-08:
Error as image . How to fix ? Thanks of watching ! 
[image]

hzeller, 2016-12-08:
You need to compile vlc from head.

ffqqtv, 2016-12-10:
[image]
I have compiled all file , but still failed . 
Thanks for watching !

hzeller, 2016-12-10:
You need to compile it from head, so you need to check it out from the git repository. The version would be something like this:
```
./vlc --version
VLC media player 3.0.0-git Vetinari (revision 2.2.0-git-10259-g56a969a)
VLC version 3.0.0-git Vetinari (2.2.0-git-10259-g56a969a)
```

Also make sure that you actually start the vlc from your compilation by providing the path where it is. With `vlc`, you just start the vlc which is in your system, but the system search path does typically not contain `./`. So you need to give `./vlc`

hzeller, 2016-12-10:
And if you do make install, then make sure to run `sudo ldconfig` afterwards to update references to shared libraries.

ffqqtv, 2016-12-12:
Thanks for replying ! i did not find version 3.0.0 of vlc . i have tried to search for 2 day , but i do not find it . Can you send to me ? thank you , so much! 
Thanks for watching !

hzeller, 2016-12-12:
As I said, compile it from git. https://wiki.videolan.org/UnixCompile/

ffqqtv, 2016-12-15:
Thanks for replying ! When make vlc , it have a error . I search very much , but  do not fix that error 
Thanks for watching ! 
[image]

hzeller, 2016-12-16:
You have to find out yourself for whatever platform you're compiling it on. Looks like you compile it on a Raspberry Pi, so you might need to find out what is specific there. The name `mmal_port_parameter_set` sounds like this has something to do with the specific multi-media libraries in RPi. So you have to have the mmal library installed. If you know someone who has experience compiling things on Unix, you can ask them to help you out.

(Also why Raspberry Pi ? Your regular Linux machine would be faster for compiling and you then can send stuff over the network to your Pi).

ffqqtv, 2016-12-17:
Thanks for replying ! i configure as ./configure --disable-lua --disable-avcodec --disable-swscale --disable-xcb --disable-alsa --disable-live555 --disable-a5 . when run
 ./vlc --vout flaschen --flaschen-display=localhost \
--flaschen-width=192 --flaschen-height=96 \
"https://www.youtube.com/watch?v=kTl-aV11rME" . Error : 
[image]
how to configure ?
Thanks for watching !

hzeller, 2016-12-17:
Read the error messages: you don't have sound and don't have a UI because you don't have a graphical user interface (because VLC still shows some window with these options).

Don't run this on the Raspberry Pi, it is too slow anyway. Run this on your normal Linux machine and send the result over the network to your Pi.

closed 2017-07-30 by hzeller