hzeller/rpi-rgb-led-matrix #253: led-image-viewer is not showing the correct image

ivarvanwooning, 2016-12-06:
Hi there! I have some trouble with getting the led-image-viewer to work on a Raspberry Pi 2. I need to display a simple animated GIF file on a 64x32 LED matrix, but when I use the command 'sudo ./led-image-viewer -r 32 -c 2 image.gif', I see [random green blocks](http://ivarvanwooning.nl/1.JPG).

All other demos work great and I have already unloaded the sound module. I have also tried re-downloading and rebuilding everything. What else could be causing this? Any help is much appreciated!

ivarvanwooning, 2016-12-06:
Also interesting: displaying a .jpg file does result in some errors (after briefly displaying the green blocks):

`Size: 64x32. Hardware gpio mapping: regular`
`Load images...`
`Trouble loading 1.jpg (Magick: Unable to open file (1.jpg) reported by magick/blob.c:2783 (OpenBlob))`
`No image could be loaded.`

hzeller, 2016-12-07:
This looks like your set-up doesn't work at all. Before you use the image viewer make sure to get your set-up working with the demo in examples-api-use/

ivarvanwooning, 2016-12-07:
Thank you for the quick answer. I started all over again, used the 'make' command and tried to run the demo in examples-api-use/. Everything executes normally, but now no LEDs light up.

It seems that only an older Adafruit repository is working.

hzeller, 2016-12-07:
Ah, so you are using an Adafruit HAT ? That one uses a different pinout and you need to give the `--led-gpio-mapping=adafruit-hat` on the command line.

ivarvanwooning, 2016-12-07:
This worked, excellent! I am indeed using an Adafruit HAT, I should've specified that ;). In case anyone else encounters the same problem, also try using the Raspbian Jessie Lite version instead of the full version. In the end, the following command worked for my 64x32 LED matrix:

`cd rpi-rgb-led-matrix/utils`
`sudo ./led-image-viewer -r 32 -c 2 --led-gpio-mapping=adafruit-hat image.gif`

hzeller, 2016-12-07:
Also, I urge everyone who has a problem to spend at least 3min reading the documentation, where this particular thing is described in length.

You get your matrix running quickly within minutes instead of days and you don't spent hours of other peoples' time which could've been used to write more libre software instead.

closed 2016-12-07 by hzeller