hzeller/rpi-rgb-led-matrix #254: How to connect to HUB12?

Czajo, 2016-12-07:
Hi, I was red a lot of information about it, but I haven't any results. 
My HUB12:
OE  1  2   A
N    3  4   B
N    5  6   F
N    7  8   S
N    9 10  L
N  11 12  R
N  13 14  F
N  15  16 F

I have no idea how to connect this for my raspberry pi 2B. Anyone can help?

hzeller, 2016-12-08:
You need to figure out what the letters on your connection diagram means. Also figure out if output enable is positive or negative - I heard that it is positive instead of negative in  the Hub75, so you might need to invert that signal.

I suspect the letters mean
  A, B : address, like in Hub75
  OE   : output enable
  S   : clock ?
  L   : latch ?
  R  : Red - the only color
  N  : probably GND

In a first step, you can probably just manually invert the OE signal with an external invertert before changing the software.
From there, you then have to figure out the mapping. With only A, B, this will be 1:4 multiplexing, so you can probably assume that this will have a multiple --led-chain long. After you have figured out the mapping where each LED shows up, you can write a Transformer that does that.

A little bit of investigating, as this library is originally meant for Hub75 panels, but the Hub12 is sufficiently close that it should be possible to get it to work.
Good luck!

Czajo, 2016-12-22:
Hi one more time. I bought inverter on chip number 7404 and revert OE signal, but my led matrix shine probably on the max power. My power suply shows overload (over 1A x multipler 2,5). Without inverter I showed the moving string but distorted (demo 2).
I have a question: On my raspi-config I2C was disabled. I changed this to enable. Maybe there will be ok?

Merry Christmas!!! :)

prunella2017, 2017-03-02:
@Czajo , hi were you able to proceed further with OE pin challenge? I too have the same issue as none of my led glows....?

MaitreDede, 2017-09-28:
Hi @Czajo 
Is your panel a [Freetronics DMD](https://www.freetronics.com.au/collections/all-products/dmd) ?
The connector looks like yours. (see their [pdf](http://cdn.shopify.com/s/files/1/0045/8932/files/DMDCON_DMDConnector.pdf?100730)).

They provide a [simple lib](https://github.com/freetronics/DMD) for arduino, a [second version](https://github.com/freetronics/DMD2), and I have made a [C# lib](https://github.com/MaitreDede/RaspberryPi.Libs/tree/master/RaspberryPi.PiGPIO.Drivers/Freetronics) for Raspberry Pi on top of [PiGPIO](http://abyz.co.uk/rpi/pigpio/). I also built [a driver](https://github.com/MaitreDede/RaspberryPi.Libs/tree/master/RaspberryPi.PiGPIO.Drivers.Drawings.ImageSharp/Dede) that uses a drawing lib (so I don't have to code them myself).

It won't replace hzeller work, but it can provide you a start point for HUB12 panels.