hzeller/rpi-rgb-led-matrix #255: Set different wait times for a sequence of images

ayman0201, 2016-12-07:
Hi,

I'm trying to use led-image-viewer to display a set of images with changing wait time in between.
What I'm trying to achieve is something like the following:
led-image-viewer image1.jpg -w3 image2.jpg -w2 image3.jpg ...

Can this be achieved using the API provided?
All I'm getting now is the wait time of the last -w argument typed, in this case -w2 for all images.

Any suggestion would be appreciated.
Thanks.

hzeller, 2016-12-08:
The API is agnostic of the image times etc, this would just be a feature of the image viewer.

It would be a good addition; you might need to experiment with the getopt() implementation though to figure out if it can deal with multiple times the same flag, as you might not see which image it refers to (maybe you can peek in the next argv?). And ues, right now the last -w argument is the one that is

If you have implemented that, I'd be interested if you send a pull request, other people might be interested in this as well.

hzeller, 2016-12-09:
Alright, added this feature.

closed 2016-12-09 by hzeller

ayman0201, 2016-12-09:
I managed to implement it earlier on today.
My version handles it as follows:
led-image-viewer img1 img2 img3 -w2,4,6 

Thanks for adding the feature :)

commit 5fdc3088 (2016-12-09): o Add a way to choose timing relevant parameters (-w, -l, -t) before
  each image.
Fixes #255