hzeller/rpi-rgb-led-matrix #261: LED RSS Newsticker

genvagula, 2016-12-26:
Hi,

Sorry if this is out of the context but thought that would be good place to start here to find help on this issue. I would like to have RSS newsticker based on this library and found even some piece of code from internet and worked a bit on this. But it has some issues and seems i'm not getting rid of them at my own. 
So my thought was that is there a chance that you or anyone over here would help and review the code and give some ideas on how to get it work. Or you would add rss newsticker sample to this library python folder. This kind of newsticker thing would be very helpful for lof of others I think.

So below is the code that i have. It works so far that i get the first bunch of headlines converted to picture files (*.ppm) and if these are scrolled then the led's will blank and new set of headlines will not be loaded.
My wish would be that if the first set of pictures are shown then it automatically gets new set of pictures and starts to show them. Or converts pics in the background so there would be always be headlines to show, even if rss feed has not been updated with new headlines meantime.

https://github.com/genvagula/led-rss-newsticker

Greatly thankful for any help. 

Regards
Gen

hzeller, 2016-12-26:
This is way to compilcated.
Don't use pictures, just use the text library provided in the C++ API and use that to write things.

genvagula, 2016-12-26:
Ok, thnx. I'll try that out.

closed 2017-01-08 by genvagula