hzeller/rpi-rgb-led-matrix #262: wiring multiple chains

roykolak, 2016-12-26:
I have connected R1, G1, B1, R2, G2, B2 to my first and second chains (two chains with two 64x64 panels). 

Question about GND, strobe, clock, OE-, A, B, C, D. It seems like ALL chains need to be wired with these connections. Is that correct? Can I just splice the one wire from the pi to both chains?

hzeller, 2016-12-26:
Yes, GND, strobe, cock, OE-, A, B, C, D (and possiblye E if you have a 64x64) need to be wired to all panels. You can do that with wires, but I'd suggest to solder up an adapter to simplify things
https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3

roykolak, 2016-12-26:
gotcha, just wanted to confirm. Thanks!

closed 2016-12-26 by roykolak