hzeller/rpi-rgb-led-matrix #264: Python bindings missing matrix arguments

danielvy, 2017-01-04:
The Python binding is missing some of the matrix arguments.

e.g:
--led-pwm-lsb-nanoseconds
--led-scan-mode
--led-slowdown-gpio

And others.

Also, it would be nice to be able to set led-gpio-mapping in Python without recompiling the library for default value.

Thanks!

hzeller, 2017-01-04:
Yes, this is a known missing feature.
Pull requests welcome.

danielvy, 2017-01-04:
I'd happily add them but my c++ is at least 20 years rusty and I don't even know where to start.

If you can give a few pointers or an example which parts needs to be modified, I can give it a try.

hzeller, 2017-01-04:
Nothing needs to be changed in the C++ part, that already is working. It is mostly about the Python adapter code here; @Saij might be able to help.

danielvy, 2017-01-08:
Ok, I gave it a shot. Everything seems to be working properly. Tested on Python 3.6 and 2.7.9 

You can find my branch here [danielvy/rpi-rgb-led-matrix](https://github.com/danielvy/rpi-rgb-led-matrix/tree/python-matrix-options).

Anyone care to take a look before I submit a pull request?

### Quick summary
* Added the structs for `RGBMatrix::Options` and `RuntimeOptions` and their Python wrapper classes
* Added `CreateMatrixFromOptions()`
* Modified the `RGBMatrix` constructor to accept optional `RGBMatrixOptions` and `RuntimeOptions` and  call `CreateMatrixFromOptions()`
* Modified `samplebase.py` to create `RGBMatrix` with `RGBMatrixOptions` and `RuntimeOptions`. Also added the more significant options as additional command line arguments
* Updated the readme.md

### Few issues:

#### # 1

This breaks backward compatibility. 

Since Python can't have multiple constructors and because how the previous constructor was written, I couldn't find any elegant way to add the options structs to the constructor. Adding them all as named arguments doesn't make sense.

If there's a better way, let me know.

#### # 2

When generating the `core.cpp`, I get the following warnings from cython:

```
 warning: core.pyx:88:39: Not all members given for struct 'Options'
 warning: core.pyx:138:54: Not all members given for struct 'RuntimeOptions'
```

Not sure why this happens. Couldn't find any meaningful reference to these warnings.

#### # 3

In order to pass a Python string to the struct's `char *hardware_mapping`, it needs to be encoded but the encoded byte array needs to be kept alive (since the options struct will be used later, after the property set call has exited). 

I solved this be adding a "private" member `cdef bytes __py_encoded_hardware_mapping` in `core.pxd` to hold the encoded bytes when the property is set. If there is more elegant solution, let me know.

- Daniel

hzeller, 2017-01-08:
For #1 - this needs to be backward compatible, otherwise people are not happy.
Could this proably just be additional optional parameters ?
  def __cinit__(self, int rows, int chains = 1, int parallel = 1, gpio_mapping = "regular", pwm_bits=11, lsb_nanoseconds=130, ... and so on)
I think then Python allows people to give these in any sequence on construction with named parameters, right ? (this is how you can tell that I don't know anything about Python :) )

I would not expose the Options and RuntimeOptions struct in Python at all to keep it simple for the user. It is somewhat an idiom needed in C/C++ because there are no named parameters.
I'd just make these additional optional named parameters in the Python constructor.

(With this approach, you probably will also not run into the issue of having to do the hardware-mapping string trick, as you set and consume the option struct just in the constructor.

Also, I would strongly like to see that you can somehow call the original rgb_matrix::ParseOptionsFromFlags() to fill in values before the user then can overrides things programmatically. This makes sure that new flags that show up automatically work also in Python, and it makes sure that the names of the flags are the same and consistent, so you don't have to add anything in samplebase.py. Be careful though, rgb_matrix::ParseOptionsFromFlags() modifies the argv array and re-arranges things in a way that the led-xyz options are already consumed. This is useful in a c-program, but might trip up the Python interpreter if the global argv changed. If it does, then we either need to make a copy (probably messy) or have an optional parameter in rgb_matrix::ParseOptionsFromFlags() that tells it to _not_ rearrange flags.

hzeller, 2017-01-08:
You could also call CreateMatrixFromFlags() which is a convenience function that combines parse and create.

hzeller, 2017-01-08:
I have added commit https://github.com/hzeller/rpi-rgb-led-matrix/commit/f1b8fe63a8f6d8af63dadcb1baac626710fde9ac that allows to choose not to re-arrange the argv[] flags.

danielvy, 2017-01-08:
> For #1 - this needs to be backward compatible, otherwise people are not happy.
> Could this proably just be additional optional parameters ?
> def cinit(self, int rows, int chains = 1, int parallel = 1, [...]

There are over 15 parameters. In most style guides (Python or other), the general rule of thumb is that of having 3-4 function parameters (named or positional) max, more than that should usually go into separate class/struct/array/dictionary/whatever which already exists here in the form of Options/RuntimeOptions, so it seemed natural to use them. 

As for `ParseOptionsFromFlags`, while nice to have something consistent in the demo, I suspect that most of the library's uses case won't use any command line at all (or have their own app logic args) and create the RGBMatrix from known set of parameters. And I am not sure how it will coexist with Python's arg parser anyway.

hzeller, 2017-01-08:
So how about you make the options named parameters after the already existing three parameters or so ? That way the constructor is backward compatible but it will have the option feature as well.

I really would like to see the standard command line options be used so that the transition is simple if someone wants to play with Python. I now made it possible to choose that the commandline parser does not mess with the sequence of the arguments, so if you someone can give it access to the original argc/argv, then everything will be working as expected.

danielvy, 2017-01-08:
> So how about you make the options named parameters after the already existing three parameters or so ? That way the constructor is backward compatible but it will have the option feature as well.

That was my initial thought too, but then I saw that the `rows` parameter in the original `RGBMatrix` constructor is not optional and since `RGBMatrix::Options` also has `rows`, it would be confusing to the user which `rows` takes precedence in case Options is provided.

hzeller, 2017-01-08:
Yeah, but think this is an ok-tradeoff to make given that we have to be backward compatible. Also, can we make rows optional after the fact now ? If people still give it as first parameter in old code it should work, but people in future code can decide to not give that parameter at all.

danielvy, 2017-01-08:
> Yeah, but think this is an ok-tradeoff to make given that we have to be backward compatible. Also, can we make rows optional after the fact now ? If people still give it as first parameter in old code it should work, but people in future code can decide to not give that parameter at all.

Yeah, I think this is doable. Then it can throw an exception if both are supplied. So the user can provide either rows (and chains/parallel) **OR** RGBMatrixOptions

Also, is there a reason to keep `RGBMatrixOptions` and `RuntimeOptions` separate on the Python side? Or can they be unified under `RGBMatrixOptions`?

hzeller, 2017-01-08:
I think they can just be merged for simplification.

I would'nt worry about throwing an exception if both values are given. The user probably will figure out that one of these wins.

danielvy, 2017-01-08:
ok then.

I will update it later tonight or tomorrow.

hzeller, 2017-01-08:
Thanks  for your work on this!

hzeller, 2017-01-25:
Fixed in https://github.com/hzeller/rpi-rgb-led-matrix/commit/e2ffd91644eb9ca36b57921c07106be1ef0d0f63

closed 2017-01-25 by hzeller