hzeller/rpi-rgb-led-matrix #267: Color Mixed

akardes5, 2017-01-09:
Hi,

The project is very nice. I did the same. But the colors are mixed.

Red -> Blue
Green -> Red
Blue -> Green

For example;
A red color image appears blue.
How can I fix?

Thank you.

hzeller, 2017-01-09:
You probably have a panel that has swapped inputs. Just wire up the R1, G1, B1 and R2, G2, B2 differently so that it matches your panel.

akardes5, 2017-01-09:
Do you know the Adafruit Hub75 R1 G1 B1?

Thank you.

hzeller, 2017-01-09:
if you are using the HAT, then better remap it in software. look in lib/hardware-mapping.c and change the p0_r1 etc to point to the gpio it should be.

akardes5, 2017-01-09:
I think this will solve the problem. Thank you so much.

prunella2017, 2017-03-19:
@akardes5 , hey, did the hub75 work for you? Thanks!.

hzeller, 2017-05-25:
There is now an option `--led-rgb-sequence` that allows to choose the order on the command line without changes to the code.
(added in https://github.com/hzeller/rpi-rgb-led-matrix/commit/efe8a77f1958c342f97a3ac7f7d0601e02c30b6a )

closed 2017-05-25 by hzeller