hzeller/rpi-rgb-led-matrix #270: Red LED Display P10

InunoTaishou, 2017-01-20:
Trying to get the rpi-rgb-led-matrix driver working with [this matrix I got off ebay.](http://www.ebay.com/itm/282289161835?_trksid=p2060353.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT)

[here's my current pinout from matrix -> pi and the chips on the board](http://imgur.com/a/ndMrT)

Now, [here's my current issue](https://youtu.be/Jqf9hJsWjqA). It just doesn't seem like the sections line up properly. The text example does something like this too, I just chose the squares demo because it was a lot more noticeable.

hzeller, 2017-01-21:
So looks like this only has 1:4 multiplexing, possibly having a display arrangement in which two longer strips with 4 LEDs are wrapped around to a total of 16 high.

So a good start is to give it the parameters `--led-rows=8 --led-chain=2`. Then you need to write a transformer by observing where pixels you set actually get mapped. Once you have written that transformer, you can register it in the library so that it can do the automatic correction.

Have a look at the CanvasTransformer described in https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/canvas.h and read the section about applying a transformer in https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use

InunoTaishou, 2017-01-21:
Thanks for the step in the right direction. Sounds like it'll take a while to figure this out completely so... I'll probably order an adafruit one for now to go forward with that I'm wanting to use it for until I get this figured out.

Out of curiosity, what matrix did you use for testing? This is a very popular library, if there was a wiki, or something, that listed the boards that would work right out of the box, or with minor tweaks, it would help. We all know the adafruit boards work.

hzeller, 2017-01-21:
There are many compatible matrix out there, they are very common. If you look for Hub75 RGB on aliexpress or ebay, you find a lot of these. And of course Adafruit.

prunella2017, 2017-02-28:
@InunoTaishou , hello....
Need your help.
I brought a hub12 led, however none of the led glows if I run the program with rasp pi 3. I definitely missing a step. Could you please share your pin-to-pin mapping between rasp and p10/hub12 led board please? This would really help.

Thanks
Sabs

InunoTaishou, 2017-03-02:
My pinout is in the first image on [the imgur post](http://imgur.com/a/ndMrT). It looks like I forgot to pinout the oe connection but I would have used the same pin from [the github post](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md).

prunella2017, 2017-03-02:
Great…thanks for the links & quick response, earlier the imgur link was not working on my computer and I had to ask for it again….sorry about that…
Quick question, what about the OE pin on the led board? Any particular reason for it to be left out? Is it because hub12 is opposite to hub75 in terms of OE signal?

This message is for the designated recipient only and may contain privileged, proprietary, or otherwise confidential information. If you have received it in error, please notify the sender immediately and delete the original. Any other use of the e-mail by you is prohibited. Where allowed by local law, electronic communications with Accenture and its affiliates, including e-mail and instant messaging (including content), may be scanned by our systems for the purposes of information security and assessment of internal compliance with Accenture policy.

www.accenture.com

InunoTaishou, 2017-03-02:
Nope, I just forgot to put the pinout in my picture. I used in on my matrix. I would have used the same pin that was used in the pinout tutorial I linked; pin 12 for OE.

prunella2017, 2017-03-02:
Right makes sense.
The vendor gave me **one** p10(aka hub12) board and also DC 5V having 20Amps, however when I tried this pinout that hzeller has mentioned in this tutorial I do not even see a single LED glowing. I am very sure, I m missing a piece.

Is there anyway before I connect my rasp device to the led board to find out that the power supply or the led board is functional?

hzeller, 2017-03-02:
Note, hub12 might be different from Hub75 - I think some signals are reversed (positive vs. negative edge etc), but I have not tried this myself yet. But it would explain why you won't see anything until you have determined what the protocol is and adapt it.

prunella2017, 2017-03-02:
Hello Henner @hzeller , good to see your response. I did go through some of the issues where you had suggested to invert.

I am bit curious to know how @InunoTaishou got the hub12 working without inverting.

prunella2017, 2017-03-02:
@hzeller ,its needless to say that many have already benefited from your code.....few ppl like me can also leverage the hub12 board config as hub12 is mostly used for outdoor commercial purpose....just a thought.... .

Haven't worked on inverting OE, I will try something and I do find out will post it back.

prunella2017, 2017-03-02:
@hzeller, Its needless to say that many have already benefited from your code.....few ppl like me can also leverage the hub12 board config as hub12 is mostly used for outdoor commercial purpose....just a thought.....

Haven't worked on inverting OE, I will try something and If I do find out then will post it back.

prunella2017, 2017-03-02:
@InunoTaishou , by the way, is your matrix running correctly?

InunoTaishou, 2017-03-02:
Mine did not. I got it to display and it tried to do the rotating square script but the square was messed up and not aligning properly on the matrix. It does turn on and try to display but I need to use CanvasTransformer to write a transformer to align everything correctly. Sounds time consuming and I just don't have the time to put into atm.

prunella2017, 2017-03-02:
I understand.

Regarding no dislpay on my board, I still couldnt find a logical ans as to why none of the led glows even though i followed the pins same as yours, and its becoming quite frustrating for me.

prunella2017, 2017-03-02:
did you invert the OE signal?

InunoTaishou, 2017-03-02:
I did not. Everything connected the way in my picture and I used a separate 5v power source (a 9v battery bucked down to 5v)

prunella2017, 2017-03-19:
Hi @hzeller , by the way, while referring the https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md#chains diagram, saw that the pins on the ribbon cable are inverted compare to the pins on the led board (panel), is this applicable for all models ?

prunella2017, 2017-03-19:
By the way @hzeller , I have tried with hub75 instead of hub12, and this is what I see now....could you please help me understand what I am missing here ....thank you.

[https://www.youtube.com/watch?v=UZpNDrSQZmQ](url)

Here are  my pics too. Pics from the same video....

And this is what I am hv been executing:
 sudo examples-api-use/demo  -D 1  --led-no-hardware-pulse --led-rows=16 examples-api-use/runtext16.ppm

[image]
[image]
[image]
[image]
[image]
[image]

hzeller, 2017-03-19:
Looks like you have one of these 1:4 multiplexing (probably out-door) displays.You might need to write a transformer to get the pixel-mapping right.

prunella2017, 2017-03-19:
Thanks for replying..Okay...is there any more details i can refer to? It would help me to kick start atleast...thanks again.....
________________________________

This message is for the designated recipient only and may contain privileged, proprietary, or otherwise confidential information. If you have received it in error, please notify the sender immediately and delete the original. Any other use of the e-mail by you is prohibited. Where allowed by local law, electronic communications with Accenture and its affiliates, including e-mail and instant messaging (including content), may be scanned by our systems for the purposes of information security and assessment of internal compliance with Accenture policy.
______________________________________________________________________________________

www.accenture.com

prunella2017, 2017-03-19:
I also thought all hub75's are outdoor.... ..Sorry my bad....
________________________________

This message is for the designated recipient only and may contain privileged, proprietary, or otherwise confidential information. If you have received it in error, please notify the sender immediately and delete the original. Any other use of the e-mail by you is prohibited. Where allowed by local law, electronic communications with Accenture and its affiliates, including e-mail and instant messaging (including content), may be scanned by our systems for the purposes of information security and assessment of internal compliance with Accenture policy.
______________________________________________________________________________________

www.accenture.com

mbates14, 2017-10-19:
When it comes to the chinese, things change daily. The Tiles I am getting now from our broker in China are 1/2 scan, and 1/4 scan. Most of the outdoor signs are 10,000 nits which is 1/2 scan. 10MM pixel pitch. This library when it was built appears to be setup mostly for indoor tiles. Everything coming out of china now are mostly outdoor 1/2 and 1/4 scan.

hzeller, 2017-10-19:
I am happy to apply pull request that can deal with the pixel mapping of these panels.

mbates14, 2017-10-19:
I have a specific panel that we get that I still need to figure out, as I want to re-purpose those for another project outside of linsn/novastar.

mbates14, 2017-10-19:
Also in case people are wondering, HUB75 is a physical connection interface standard. Anything that is HUB75 insures that the pins will match and can be driven physically the same. Electrically, obviously they are all different. HUB12 is monochrome.

AbeeB, 2019-03-27:
friends can anyone tell me the correct procedure for interfacing arduino megha and p10 hub12 LED