hzeller/rpi-rgb-led-matrix #274: LEDs are trembling too much

akardes5, 2017-01-29:
Hello, the project is very nice. I did the project and it works beautifully. I have a problem. I used the P4 rgb led matrix. LEDs are trembling too much. How can i solve it? Thank you.

hzeller, 2017-01-29:
You have not described any of the circumstances, but without information, it is hard to tell what needs solving.

  * How many panels do you have connected in a chain ? What is the panel size ? What are the command line flags you give to the program ?
  * What is the Pi you are using ? Pi1, Pi2 or 3 ?
  * Do you build an active adapter from this project or use a HAT from Adafruit ?
  * What are the symptoms ? "Trembling" is a bit generic. Do you see random LEDs light up that shouldn't, or do you see brightness glitches ?

akardes5, 2017-01-29:
I'm sorry you're right. I am using raspberry pi 3. I use a 5v 15A adapter. I use adafruit hat. I use 2 p4 rgb matrix panels. The white light is trembling. Animation is very trembling. I do not have the information in the code area. Thank you.

hzeller, 2017-01-29:
You first should start with using PWM with the Adafruit HAT
  https://github.com/hzeller/rpi-rgb-led-matrix#improving-flicker

Then, make sure not not run else anything CPU intensive on the Pi.

Use the `isolcpus` configuration as found in the README.

akardes5, 2017-01-29:
Thank you. I will try and give information.

akardes5, 2017-01-31:
Hello,

I did not succeed :(
Raspberry Pi programming is also very new. For being able to understand it as Tama. Could you tell me a little more detail? What kind of code should I write? What should I change in the "rpi-rgb-led-matrix" folder? Please help me.

Thank you.

hzeller, 2017-01-31:
You don't have to change anything in the folder.
I have elaborated a lot in the README, I expect you to read it first.

For the PWM change, you need to solder a bridge on the Adafruit HAT and then run the examples with `--led-gpio-mapping=adafruit-hat-pwm`.

All the other changes are as well just different options. Please read the README

akardes5, 2017-01-31:
--led-gpio-mapping = adafruit-hat-pwm. I do not know how to run the examples. would you tell? Thank you.

hzeller, 2017-01-31:
Please make an effort to please read the README first before sending messages to at least 75 people.

closed 2017-07-30 by hzeller