hzeller/rpi-rgb-led-matrix #275: Error when running python runtext.py

popeyesfx, 2017-02-04:
Hi
I am trying to use the python lib, but get the following error:
sudo ./runtext.py
Traceback (most recent call last):
  File "./runtext.py", line 3, in <module>
    from samplebase import SampleBase
  File "/home/pi/pywork/pibanner/rpi-rgb-led-matrix/python/samples/samplebase.py", line 7, in <module>
    from rgbmatrix import RGBMatrix, RGBMatrixOptions
  File "/home/pi/pywork/pibanner/rpi-rgb-led-matrix/python/rgbmatrix/__init__.py", line 7, in <module>
    from .core import RGBMatrix, FrameCanvas, RGBMatrixOptions
ImportError: No module named core

Think i might have missed something, any help would be appreciated.

thanks

jack

popeyesfx, 2017-02-04:
I also noticed a warning when building:
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
not sure if this has any bearing on the above issue

hzeller, 2017-02-04:
don't worry about the warning, that is because the generated python compiler thing adds that.

Make sure to `sudo make install` in the python directory before you can expect the python stuff to work.

popeyesfx, 2017-02-06:
Thanks for the answer can i just clarify a couple of things

I have the rip matrix software installed at
home/pi/pywork/rpi-rgb-led-matrix

I ran the build and install from this directory as the instructions in the python folder said to do it from the root, is that correct or should i have run

sudo apt-get update && sudo apt-get install python3-dev python3-pillow -y
make build-python PYTHON=$(which python3)
sudo make install-python PYTHON=$(which python3)

in the python folder.

I'm new to make files so wasn't sure if i should be replacing (which python3 )with something

thanks

jack

popeyesfx, 2017-02-06:
I ran:
pi@pibanner:~/pywork/pibanner/rpi-rgb-led-matrix/python $ make build-python PYTHON=$(which python3)

and got:
make -C ../lib
make[1]: Entering directory '/home/pi/pywork/pibanner/rpi-rgb-led-matrix/lib'
make[1]: Leaving directory '/home/pi/pywork/pibanner/rpi-rgb-led-matrix/lib'
/usr/bin/python3 setup.py build --build-lib .
running build
running build_py
running build_ext

then ran:
pi@pibanner:~/pywork/pibanner/rpi-rgb-led-matrix/python $ sudo make install-python PYTHON=$(which python3)

and got:
/usr/bin/python3 setup.py install
running install
running build
running build_py
running build_ext
running install_lib
running install_egg_info
Removing /usr/local/lib/python3.4/dist-packages/rgbmatrix-0.0.1.egg-info
Writing /usr/local/lib/python3.4/dist-packages/rgbmatrix-0.0.1.egg-info

then tried:
pi@pibanner:~/pywork/pibanner/rpi-rgb-led-matrix/python $ cd samples
pi@pibanner:~/pywork/pibanner/rpi-rgb-led-matrix/python/samples $ sudo ./runtext.py

and got the fail:
Traceback (most recent call last):
  File "./runtext.py", line 3, in <module>
    from samplebase import SampleBase
  File "/home/pi/pywork/pibanner/rpi-rgb-led-matrix/python/samples/samplebase.py", line 7, in <module>
    from rgbmatrix import RGBMatrix, RGBMatrixOptions
  File "/home/pi/pywork/pibanner/rpi-rgb-led-matrix/python/rgbmatrix/__init__.py", line 7, in <module>
    from .core import RGBMatrix, FrameCanvas, RGBMatrixOptions
ImportError: No module named core
pi@pibanner:~/pywork/pibanner/rpi-rgb-led-matrix/python/samples $ 

not sure what I'm missing?

cheers

popeyesfx, 2017-02-06:
Just did a clean and reinstall and noticed that when i do:

sudo apt-get update && sudo apt-get install python3-dev python3-pillow -y

it selects oil and not pillow to install, would this effect the results of the build

cheers

hzeller, 2017-02-06:
Looks like you install things for python3, but then run python2.7 ? Runtext will use whatever default python there is, so that would be 2.7 typically.

So you either also have to compile/install the Python library for 2.7, or use Python3 in your runtext.

(Note that Python3 is a little slower, so only use if needed for your application)

popeyesfx, 2017-02-08:
Thanks for the reply
I will retry it and see how it goes

popeyesfx, 2017-02-09:
Thanks for the help, i got it working now. 
I do have another question, I would like to run the scroll code without the press control c to exit. the reason for this is i want to be able to change text on the fly.

Im trying to get it to work from a php web page. I exec a small script called message which then uses the os to call runtext. message works fine from the command line but not from the php file. 

as below. 
php code
 <?php
            if ($_GET){
            $temp = exec("sudo python3 /home/pi/pywork/pibanner/rpi-rgb-led-matrix/python/samples/message.py ".$_GET[data]);
            }
        ?>

message.py code
def main():
    if len(sys.argv) > 1:
        msg = sys.argv[1]
        os.system('sudo python3 /home/pi/pywork/pibanner/rpi-rgb-led-matrix/python/samples/runtext.py  -r16 --led-no-hardware-pulse LED_NO_HARDWARE_PULSE -t' +msg)
        print(msg)

if __name__ =="__main__":
    main()

popeyesfx, 2017-02-09:
any help with the above would be appreciated 
thanks

hzeller, 2017-02-10:
Of course, you can so os.system() and then kill the program later (with whatever the kill call is in PHP). But first, this is dangerous (you essentially have to run the PHP stuff as root). Also, it is fragile because you essentially have to use PIDs as communication mechanism. And it will create glitches when the LEDs go off and are resynchronized.

Instead, you should just write a little TCP server in Python that receives new strings and then displays that whenever it changes. From your other PHP program, you then send a message to that socket. This is probably a handful of lines of code and the right way to do it.

closed 2017-02-10 by hzeller

popeyesfx, 2017-02-10:
Thanks for all your help, it is very much appreciated