hzeller/rpi-rgb-led-matrix #276: Trembling leds while using Ethernet

currentlysober, 2017-02-06:
Dear All,

I'm not sure whether someone did notice the problem with trembling/blinking leds while using Ethernet connection. My program displays information from the txt file saved on RPi. However, when my computer sends the new txt file to the RPi, (I guess) cpu is lit bit overloaded (in a very short period of time) and leds blink a few times. All in all, the display is not stable. And when I send some files constantly e.g. once per 5 seconds, it blinks/trembles nearly all the time.

Of course I read README, I tried to use 'isolcpus=3' solution but it is not helping. I tried also isolcpus=1,2,3.
Moreover, I found this post today:
https://www.raspberrypi.org/forums/viewtopic.php?f=33&t=153029
but I do not know how to implement it and if this can be our solution.

I am using RPi 3. 
Well, maybe this is some HW bug/problem of losing electric voltage while using Ethernet port..? I hope not :)
You know, we can slow down the performance of Ethernet, the most important is to have stable rgb-led-matrix library performance.

I'll be really grateful for your help.
Regards!

maddogs, 2017-02-06:
I don't have this problem.

In our setup, we have two displays (32x64) running on Pi2's using an active adapter.  One connected by WiFi the other Ethernet.

Ours read the text files from a server every 2.5 seconds. The difference between our setup to yours is we read from the server instead of saving the files to the Pi. Have you tried it this way round?

hzeller, 2017-02-06:
`isolcpus=3` is correct (as the display thread attempts to be tied to that core), but if other things are going on in the system that overload the memory bus, you still get flicker.

The Pi is a simple system - there is not much isolation of memory busses and other things in the Pi that make the hardware wait - they will influence other things going on. When that happens, you can see a flicker in the matrix display (e.g. already running 'top' via ssh can make your display flicker regularly whenever it updates the stats).

Usually, sending things via Ethernet works great (I am doing that with the FlaschenTaschen protocol and watch movies with it without any flicker), but as @maddogs said, don't try to save things on the file-system (you are accessing a slow SD card), just write a server that receives your data, keeps it in RAM and displays it from there. Or try reading it remotely.

Use a simple network protocol like a permanent open TCP connection or simply a UDP packet whenever there is an update. Don't use ssh/scp (there is a lot of CPU overhead just to get the encryption keys set-up for each connection. That will likely cause flicker as well. If you do it, make sure to keep the connection open: have a running ssh session in which you port-foward things. But then again, you just have a secure version of a socket, so typically a simpler manual TCP or UDP connection would be as good).

If you happen to use Python, consider doing it in C++ instead for less overhead.

hzeller, 2018-03-17:
Check out the new compile-time option in `FIXED_FRAME_MICROSECONDS` in [lib/Makefile](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/Makefile#L105). It will allow to time-buffer glitches created by the rest of the system.