hzeller/rpi-rgb-led-matrix #278: Multiple pixel lit

agoesssss, 2017-02-07:
Hi there, is there anyone ever experienced multiple vertical pixel lit when trying to SetPixel single pixel. I'm trying to turn on single pixel but I always got 4 vertical pixel lines. 

Can anyone address this issue? 

Thank you

agoesssss, 2017-02-07:
[image]

That is what I got when doing SetPixel(0, 0, 255, 255, 255);

hzeller, 2017-02-07:
This is interesting, does that happen on any pixel, eg. if you choose one in row 2, does it go from row 2 to 6 ?

This might be a slightly different panel control. CS sounds suspicious (this abbreviation usually means 'Chip Select', used to tell a subsystem that whatever data it gets is supposed for it).
It can be detrimental if that is floating (i.e. not connected). Try to set that to 5V in one experiment and to GND in another (I expect the panel to not show anything in one or the other experiment).

Can you take a good readable picture of the connector on the back of the panel and post here ?

It could also be a different issue in which it does multiplexing a bit differently, which we have to find out.

agoesssss, 2017-02-08:
I guess that's nothing to do with the chip select that I also suspect earlier, because when I tried to pulse that CS then my system freeze and nothing happened when I set to ground, and then I realized that CS is shorted to the ground. I guess there's issues with multiplexing

these is what I got when doing
matrix_->SetPixel(0, 0, 255, 0, 0);
matrix_->SetPixel(1, 1, 0, 255, 255);
matrix_->SetPixel(2, 2, 0, 0, 255);
[image]

matrix_->SetPixel(1, 0, 255, 0, 0);
matrix_->SetPixel(2, 0, 0, 255, 255);
matrix_->SetPixel(3, 0, 0, 0, 255);
[image]

matrix_->SetPixel(0, 1, 255, 0, 0);
matrix_->SetPixel(0, 2, 0, 255, 255);
matrix_->SetPixel(0, 3, 0, 0, 255);
[image]

Also attached below the pinout
[image]

Thank you for your support

agoesssss, 2017-02-08:
Attached below the video while doing sweep test using the option "--led-rows=16 --led-chain=2"

`for( int sweepY = 0; sweepY < 16; sweepY = sweepY + 1 ){
  for( int sweepX = 0; sweepX < 64; sweepX = sweepX + 1 ){
    matrix_->SetPixel(sweepX, sweepY, 255, 255, 255);
    usleep(10000);
  }
}
`
[VID_20170208_091053.zip](https://github.com/hzeller/rpi-rgb-led-matrix/files/759647/VID_20170208_091053.zip)

agoesssss, 2017-02-08:
This is the product website http://www.qiangliled.com/products-63.html

Arvindace, 2017-06-21:
Did we figure out a solution to this problem. I am using the same panel and facing the same issue. please help. thanks!

vapiper, 2018-01-12:
Hello Henner!
I bought the same matrix and got the same result as described above.
[image]
I had LED screen controller Onbon BX-5Q1+. The matrix is working correctly with this controller. Here is a chart from HUB75 of the controller (2 matriсes 32x16 in 1 chain, CH0..CH5 color bits):
[image]

And here is the chart from Rpi+Adafruit Hat (rows=16, chain=1):
[image]

The pin CS is connected to GND  inside the matrix ( When I applied +5V I got a short circuit).
If I understand correctly, in the matrix multeplexing chip is missing and rows are selected directly by the lines A, B, C, D.
This weekend I will try to make changes in methods `Framebuffer::InitGPIO` and ` Framebuffer::DumpToMatrix`.

vapiper, 2018-01-13:
My assumption was correct! Now matrix Qiangli X10 is working properly.
This is a graph after making changes to the code were made:
[image]

hzeller, 2018-02-17:
So looks like @vapiper has fixed this one with the #509 pull request resulting in commit https://github.com/hzeller/rpi-rgb-led-matrix/commit/b6892aa85666c21e7e9a9392285a278f795755e5

closed 2018-02-17 by hzeller