hzeller/rpi-rgb-led-matrix #279: Trouble in Installation of Supplementary Fonts

Acqua-Alta, 2017-02-09:
Hi, I'm troubled with installation of supplementary font, although I read "rpi-rgb-led-matrix/fonts/README".

Firstly, I can't find the directory of "submission".
Should I execute "mkdir submission", and continue to following command?

```
cd submission/
make
su
mv -b *.pcf.gz /usr/lib/X11/fonts/misc/
cd /usr/lib/X11/fonts/misc/
mkfontdir
xset fp rehash
```

hzeller, 2017-02-09:
the README in that directory is referring to the use of these fonts in your system. But you don't need that for the matrix.

Acqua-Alta, 2017-02-09:
OK, so I'll just skip this process.
Then, how to use the supplementary fonts?
I downloaded the supplementary packages from "http://www.cl.cam.ac.uk/~mgk25/download/ucs-fonts-asian.tar.gz".
Then I unpacked at the directory of "rpi-rgb-matrix/fonts".
And I executed 
```
sudo ./text-example -f ../fonts/12x13ja.bdf
```
But nothing indicated on the led matrix.

Acqua-Alta, 2017-02-09:
Sorry, I succeeded.
I just need to type texts on command line.

closed 2017-02-09 by hzeller

Acqua-Alta, 2017-02-09:
Then would you tell me how to indicate characters with scrolling in python using supplementary fonts?

hzeller, 2017-02-09:
Just look at the code of the runtext.py example, there the font is chosen. Look for `LoadFont`.

Acqua-Alta, 2017-02-09:
I've done it!!
I greatly appreciate for your quick reply and precise pointing out, and of course, for your program!

```
sudo nano runtext.py
```

Before
```
...   
   def run(self):
        offscreen_canvas = self.matrix.CreateFrameCanvas()
        font = graphics.Font()
        font.LoadFont("../../fonts/7x13.bdf")
...
```

After
```
...
    def run(self):
        offscreen_canvas = self.matrix.CreateFrameCanvas()
        font = graphics.Font()
        font.LoadFont("../../fonts/12x13ja.bdf")
...
```

And it work out as I hope!

Acqua-Alta, 2017-02-09:
Sorry, can I ask one more question?
This program designates the encode of font file, 'utf-8'.
Suppose that I use .bdf file encoded not  'utf-8' but 'shift-jis' or 'JIS X 0208', where should be changed?
I wish I could find .bdf file encoded with utf-8, but I can only find .ttf or encoded by 'JIS' file.

hzeller, 2017-02-09:
1) You could manually recode the codepoints from JIS to UTF-8 (you need to change the ENCODING from the JIS encoding to whatever the codepoint would be in UTF-8). This can be tedious though, but I suspect that font-editors can provide that feature as well - load the BDF file and write it out with UTF-8 encoding.

2) Look at the code. `graphics.DrawText()` gets the y-coordinate as 4th parameter. So if you call DrawText() multiple times with different y-coordinates, you get multiple lines of text.

Acqua-Alta, 2017-02-17:
Solved : 
1. I changed encoding of .bdf file with FontForge in ubuntu, from Shift-JIS to UTF-8 (ISO 10646-1)

Solved : 
2. I understood your description and now I can display strings in two rows.

Thank you.

But may I ask one more question?
Although I feel really sorry about asking you again and again.

That's following:

The font files which you suggested for Chinese, Japanese, Korean worked well.
But, some of kanji fonts was terrible.
Concretely to say, part of kanji as simple as alphabet, was well displayed.
But complex kanji were the state of being crashed.
So I decided to use alter kanji fonts, shinonome font.
In  http://openlab.ring.gr.jp/efont/dist/shinonome/
shinonome0.9.11p1... includes the font.

I'll show figure as sample. (shinmk14.bdf)

This is default.
[image]

It worked such as this.
[image]

Another sample
[image]

So, then I shifted for few bits.

Then, it worked well.
[image]

[image]

However, when I tried to display two characters sequently, it worked as following.
[image]

What causes this odd phenomenon?

Where should I change?

Additionaly, I thought I could solve this problem if I just shift bits about all characters anyway.

But, some of fonts didn't require to be shifted such as the following character.

[image]

[image]

How should I handle these difference?