hzeller/rpi-rgb-led-matrix #280: List of projects using this library?

Gymnae, 2017-02-09:
First of: Thank you very much for this great library. I'm finalizing a 64x64 RGB shadow box project for multiple uses.
My initial ideas was to simply display images, but with the rpi3 powering the display and this library being in python, I wondered if there's already a list of deviated projects using this library.

Like clients, server apps, weather scripts etc. I'm aware that there's pixelpusher or flaschentaschen as higher level apis, but couldn't find control clients for web or android. Yes, consumption ready scripts, because I'm no coder :-(

I'm coming from PIXEL 2.5 - which can also address my panels, but limits the color palette to 256 and 8bit. It has a very nice mobile app for controlling, something that would be great to have.

hzeller, 2017-02-09:
There are lots of projects that use this library and do all kinds of interesting things - from art to weather display, from truck weight scale to news ticker, from sports score counter to video wall. I don't think there is a comprehensive list though, and many people don't bother putting their projects on github.

Designing simple protocols (or use FlaschenTaschen) and writing clients for web or Android should not be a big problem, there are certainly coders around you that can help.

For PixelPusher, there is a LedLABs mobile application, but I think it only works on iOS.

Gymnae, 2017-02-18:
Thanks,
I will see what I can find available and dive back into learning python, codeacademy offers a nice tutorial and thanks to the now running panel, there's instant gratification, too :)

aycgit, 2017-03-02:
I would love a list like that! Even if it is not big, it would be great for learning. If someone can do it please! Even if you dont wan't to make one please give me project names/links so i can make one

@hzeller could you give me some links for those projects.

closed 2017-07-30 by hzeller