hzeller/rpi-rgb-led-matrix #282: question on api documentation

e1bilava, 2017-02-12:
Hi,

My question is maybe dumb but i'm new on raspberry, git and having no knowledge on C++/python but i have follow the readme and after installlation and wiring the sample work perfectly. but i got a question where can i find the doc on the api? for exemple on the python sample runtext.py how can i have info on : graphics.DrawText(offscreen_canvas, font, pos, 10, textColor, my_text)
or how can i know the method i can use on graphics class ?

Thanks you

hzeller, 2017-02-12:
The Python API mostly just mirrors the main C++ API, so for documentation just read the corresponding c++ header, e.g. [graphics.h](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/graphics.h).

e1bilava, 2017-02-13:
Ok thank you.

closed 2017-02-13 by e1bilava