hzeller/rpi-rgb-led-matrix #284: Pi3 Compatibility issue?

janetleekim, 2017-02-13:
Hi Henner,
I found your repo through this [tutorial.](https://learn.adafruit.com/connecting-a-16x32-rgb-led-matrix-panel-to-a-raspberry-pi?view=all) 

I purchased all of the materials in the tutorial, except I am using a Raspbery Pi 3 Model B (1GB Ram). When I got to this part: 
`sudo ./led-matrix 1 runtext.ppm`

I got the message saying the program I was using is out of date, so I downloaded the most current repo. I ended up doing this 3 or 4 times, each time trying to run the updated code (./demo etc) but it still wasn't working.
 
Then I found [this](https://forums.adafruit.com/viewtopic.php?f=47&t=83219&hilit=led+board+16x32+RGB+LED) on the Adafruit forum, so I ran the Experimental Python code at the bottom of the aforementioned tutorial, and the RGB display worked. So I know that the hardware is working. 

Is it possible that the latest version of this repo is incompatible with the Pi3? Or could there be another reason why I might not be able to use this library? Any help or advice would be appreciated, as I very much would like to be able to use this amazing library. Thanks in advance!

hzeller, 2017-02-14:
Don't read the Adafruit tutorial, it is hopelessly out of date and they don't care to make an effort to update it. The current wiring documentation you find here: https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md

>  got the message saying the program I was using is out of date,

Read the message carefully: It directs you to go to the directory `examples-api-use/` and run `./demo`. So you already have everything properly checked out, you just need to go to the sub-directory.

This version is perfectly compatible with any Pi, including the Pi 3.

janetleekim, 2017-02-14:
Thanks! Rewiring according to your diagram did the trick although I'm seeing lines in the LED - would that be because of the 3.3V to 5V shift issue?

hzeller, 2017-02-14:
Do you have a picture of the line artifacts?

janetleekim, 2017-02-15:
Please see attached. That said, this library is super cool!

[image]
[image]
[image]
[image]
[image]
[image]

hzeller, 2017-02-15:
Yes, these kind of artifacts can be due to missing level shifters.

I'd try for now to further slow down the data rate with an option `--led-slowdown-gpio=2` (or even higher number) and see if that makes a difference.

Sometimes improving the stability of the incoming power can already help. In particular if you power the panels through a longer cable, the voltage can vary at the terminal a lot, as the panel draws very short spike currents, making the digital inputs more finicky. So a 2200 microfarad (ideally low ESR) electrolytic capacitor right across the power terminals at the matrix can already help.

Many panels do need level shifters as they won't work reliably with 3.3V logic levels; it depends on their input stage. Also it is a good idea in general as it better protects.the Raspberry Pi GPIOs when some accidental short happens.

I suggest to build up this circuit, but it requires some mild SMD soldering:
https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3

janetleekim, 2017-02-15:
Gotcha -- thanks for the thorough explanation! It is super helpful to a n00b like me (although I am very eager to learn more about hardware!). as the Adafruit tutorial recommended, I bought a [74AHCT125 quad level shifter](https://www.adafruit.com/product/1787) along with a breadboard, but haven't been able to find an easy-to-follow diagram about how to connect the shifter to the LED and Pi. In any case, thanks again!

hzeller, 2017-05-12:
did you get it to work ?

janetleekim, 2017-05-12:
We ended up optimizing the colors for the glitches, i.e. using a white background with red text, so you couldn't see the glitches. (see image below -- had to cover with a plastic bag because it was blindingly bright) 

[image]

Thanks for all your help! I am no longer at the company (used my professional development budget to buy the hardware, so it stayed with them) but hopefully will either revisit the project or move on to a different one at some point!

hzeller, 2017-05-13:
Cool! Happy future LED project!

closed 2017-05-13 by hzeller