hzeller/rpi-rgb-led-matrix #285: Pi 3 Compute Module

andygrace, 2017-02-15:
Been lurking for a long while now but want to introduce myself and thank Henner for so much work in this and dealing with some high end, but also some pretty low end comments.

This is more of a survey than anything else.  I've used the library quite a bit recently and it's working great. I especially like the transformations and the virtual canvas. 

I've been knocking around an idea in my head to use the new Pi 3 Compute module and see if we can push quite a significant number of chains out the extra GPIOs, rather than just the single chain on Adafruit or three in parallel via the nice little OshPark buffer board Henner has designed and donated to the community. 

If I strip back almost everything in a distro to more or less the kernel, GPIO and networking primitives, then allocate maybe 16 parallel chains, I'm wondering how far we can push this thing. I'm even prepared to look at the sparsely documented VPUs in the VideoCore GPU to accelerate some of the basic maths routines.  

It'd be a pretty big side project - does anyone think it's worth it?

Cheers
Andy

hzeller, 2017-02-15:
I have thought about that and it is on my TODO list.

I've briefly looked into it and with the additional available GPIOs, you get about two more chains I think (maybe three) (Also I have another idea to maybe double that.) But 16 parallel chains won't be possible with the available pins.

GPU won't help much for pushing out data, but it might be possible to use to generate pixel-data.

So yeah, it is on the TODO list, whenever some free weekend permits.

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828