hzeller/rpi-rgb-led-matrix #286: Error with python

Abhigyanabhijeet, 2017-02-15:
Hi, while running python samples I am stuck here:

`sudo` ./runtext.py
`Traceback` (most recent call last):
```
File "./runtext.py", line 3, in 
from samplebase import SampleBase
File "/home/pi/pywork/pibanner/rpi-rgb-led-matrix/python/samples/samplebase.py", line 7, in 
from rgbmatrix import RGBMatrix, RGBMatrixOptions
File "/home/pi/pywork/pibanner/rpi-rgb-led-matrix/python/rgbmatrix/init.py", line 7, in 
from .core import RGBMatrix, FrameCanvas, RGBMatrixOptions
ImportError: No module named core
```
I have read previous issues related to this kind I have tried using ( @Saij  ) https://github.com/hzeller/rpi-rgb-led-matrix/issues/121. and this doesn't fix my problem, I have Jessi (complete ) on my Pi so I don't need to install python again.

Thank you 
Abhijeet

hzeller, 2017-02-15:
you need to install the rpi rgb matrix Python library first. Please read the Readme in the python/ subdirectory.

Abhigyanabhijeet, 2017-02-15:
So I followed these steps : 
```
sudo apt-get update && sudo apt-get install python3-dev python3-pillow -y
make build-python PYTHON=$(which python3)
sudo make install-python PYTHON=$(which python3)
```
...and I am still getting the same error.

hzeller, 2017-02-15:
but in the above example, you run runtext with the default Python, not python3.

Abhigyanabhijeet, 2017-02-15:
Thanks a lot. It worked. Thank you so much

closed 2017-02-15 by Abhigyanabhijeet

OllmGrb, 2019-10-09:
Hi, im having the same 'no module named core' error. I didnt understand what you meant by running runtext with the default python not python 3. Could you please clarify? thanks

hzeller, 2019-10-09:
python 2.x is the default on these Pis as far as I know. So if you invoke python , it will be that.

If you compile the library with Python3, you also have to invoke your program with python3, not the default one.

OllmGrb, 2019-10-09:
Hi, sorry im still not grasping the concept. 
Followed these steps:
sudo apt-get update && sudo apt-get install python3-dev python3-pillow -y
make build-python PYTHON=$(which python3)
sudo make install-python PYTHON=$(which python3)

and ran the program as follows
sudo ./runtext.py --led-rows=64 --led-cols=64

when the core error comes up. Sorry for not understanding, new to this thanks

OllmGrb, 2019-10-09:
Tried the make build-python with python3.7 instead, all to no avail

hzeller, 2019-10-09:
You have to invoke runner.py with python3.
```
  sudo python3 ./runner.py ....
```
... or modify runner.py to invoke python3.

hzeller, 2019-10-09:
(I suggest in general to use C++, it is much easier than using Python)

OllmGrb, 2019-10-09:
Only familiar with c and python, and am pressed for time to learn a new language

hzeller, 2019-10-09:
There is also a C-binding in `include/led-matrix-c.h` (see [c-example.c](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/examples-api-use/c-example.c)).

But invoking the binary with Python3 should solve the python problem.

OllmGrb, 2019-10-09:
thanks. quick question, cant locate the runner.py

OllmGrb, 2019-10-09:
python3: can't open file './runner.py': [Errno 2] No such file or directory
 got this as a result

OllmGrb, 2019-10-09:
Oh damn lol my bad runtext instead of runner. 
On another note, im getting alot of other pixels switching on apart from the hello world text. onboard sound is disabled, as well as 1-wire. any ideas? followed your wiring diagram to spec

hzeller, 2019-10-09:
Look at the troubleshooting section for all kinds of things to look at. You probably want to increase gpio slowdown.
Also, if you have connected it to the Pi directly without level-shifters your panel will only work if it accepts 3.3V logic.

OllmGrb, 2019-10-09:
Ok kind of solved the issue. I was using the ready-made connectors that come with the screens. I changed to jumper wires and all functioned correctly. are the grey connectors only useful to chain screens?

hzeller, 2019-10-09:
The gray connectors are working here as well, but you want the shortest possible wiring if you connect it directly to the Pi as the Pi barely will be able to drive these. So if you changed the wiring that probably helped that problem.