hzeller/rpi-rgb-led-matrix #287: 32x32 led matrix dual display

Abhigyanabhijeet, 2017-02-15:
Hello Again,

I am using these led matrix plates and the problem with them is 1st and 9th line is internally connected so even if i write a sentence in first row it gets copied in the next row. As you can observe in picture Ada is repeated twice but I coded just for one row.
[image]

Is there any library change which can fix this? Thanks for responding to previous post so quickly.  

Thank you 
Abhijeet

hzeller, 2017-02-15:
* Do you have a picture of your Hub75 connector on your matrix ?
 * What command line parameters do you give the programs you run ?
 * do you know the multiplexing of your matrix from its documentation?

Abhigyanabhijeet, 2017-02-18:
Yes sir!
I just acquired all data , here it is: 
A) image of Hub 75 

[image] 

In place of "CLK" its printed "S". "LAT" is replaced bu "L" and "GND" at bottom right is replaced by "Dg"

B) command line is exactly this ->  $ sudo ./demo -D 1 runtext.ppm

C) No documentation was provided by seller. But we have "Q6A-SL(M4)(1X2)192X192H" Board name mentioned

Abhigyanabhijeet, 2017-02-18:
Board Name is "q6d8v5"

hzeller, 2017-07-30:
Sounds like you might have a 1:8 multiplexing in a way that the second and forth stripe is just the longer part of the rest.
Try `--led-rows=16 --led-chain=2` and observe how things look.

closed 2017-07-30 by hzeller