hzeller/rpi-rgb-led-matrix #289: Wrong color.

nutthakun, 2017-02-19:
Hi i use 32x64 led panel connect 2 parallel by jumper wire. In parallel2 color is wrong at row 1-16 but 17-32 is work. 
help me please. 

https://goo.gl/photos/57s8pkximVGq3wJn7
https://goo.gl/photos/57s8pkximVGq3wJn7

Thank you.

nutthakun, 2017-02-19:
I use raspi 2 and raspi 3 it same problem.

hzeller, 2017-02-19:
This symptom looks like you might need 3.3V -> 5V logic level converter.

nutthakun, 2017-02-20:
Thank you for answer.
Use this for all gpio of row2 or use R1 G1 B1 only ?
http://112.124.9.237/images/0E92A13C47924EE889EDABC0336B64F5.jpg

hzeller, 2017-06-04:
You need it for all data lines. I usually use 74HCT245.
See https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3

closed 2017-06-04 by hzeller