hzeller/rpi-rgb-led-matrix #292: Connecting to a 64x32 LED matrix

himanshu1691, 2017-03-14:
I have a 64x32 Matrix and I want to connect it to my Raspberry Pi3 using the female to female jumper cables. I am fairly new to this and I looked at few links to find out the correct connection:

1. [adafruit](https://learn.adafruit.com/assets/23768)
2. [rpi-rgb-led-matrix](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md)

The connections mentioned in the second link are quite different from the first link. I do understand that the connections on the 64x32 matrix are different from the smaller one's but I am still not sure.

If it helps here is a link that shows how my LED matrix looks like: [LED back image](https://images-na.ssl-images-amazon.com/images/I/61vlp7v8-HL._SL1000_.jpg)
It has no labels on it whatsoever.
I just don't want to blow up the matrix and hence a little bit hesitant in trying it out.
Please help me out

hzeller, 2017-03-14:
If you are using this library, don't use the Adafruit description - this is a very old tutorial describing an old version of this libarary and it is totally out of date.

The 64x32 display connection is exactly the same as the 32x32. You just run it as if it was two in a chain, with the parameter `--led-chain=2`

On the panel picture, the input is probably the one we see on the left (Reason: there are arrows pointing up, and connections go from left to right in that case. typically).

It is hard to blow up the Matrix, don't worry. If you keep the plus and the minus properly connected, this will be fine.

himanshu1691, 2017-03-14:
@hzeller Thanks a lot.
Just a small doubt. According to my understanding of the wiring connection table, the connections marked with [3] [2] are the connections used when I am using 3 matrices in chain. So if I am using only 1 panel, I only have to connect the one's that have [1] mentioned and the A,B,C,D lines?
I apologize for my silly questions.

hzeller, 2017-03-14:
... and strobe, clock and ~OE. All the ones with a :smile: ￼symbol.

But yes, the [2] and [3], you don't have to connect.

himanshu1691, 2017-03-14:
Thanks  lot for all the help :)

closed 2017-03-14 by himanshu1691