hzeller/rpi-rgb-led-matrix #293: Which RGB 64*32 Matrix

Einhornkot, 2017-03-14:
Hi,

I want to build a LED-Matrix for my living room.

I searched a little bit and found this one:
https://de.aliexpress.com/item/256-128mm-64-32-pixels-1-16-Scan-Indoor-3in1-SMD-RGB-full-color-P4-led/32514844235.html?spm=2114.13010208.99999999.304.C85bTM

How can I see if this will work properly with this libary ?

Philipp

iammenotyou, 2017-03-21:
I purchased https://www.aliexpress.com/item/256-128mm-64-32-pixels-1-16-Scan-Indoor-3in1-SMD-RGB-full-color-P4-led/32514844235.html   And it works fine. :)

Einhornkot, 2017-03-26:
Thank you, now I can buy free from worry the matrix.

And how much Ampere has your Power supply ?

Philipp :)

iammenotyou, 2017-03-26:
I'm using adafruit's 5V 4A power supply.  I was able to use it without a 3.3v to 5v converter (but supplies do change so you might need it), and depending on what i'm displaying it's using 2.2-3.5 w.   If nothing is on the display, it is using 1w.

Einhornkot, 2017-05-08:
My LED-Matrix arrived today and now I dont know if I need a level shifter at the GPIO-Pins.

Die you connect the Matrix directly to the Raspberry?

iammenotyou, 2017-05-08:
Depends on your particular board.  Some require it, some don't.   I would suggest trying it without.  If the demo doesn't run properly, then purchase the level shifter.

Assuming your only using one board, follow the chart at: (match the happy faces)
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md

Good luck!

closed 2017-06-04 by hzeller