hzeller/rpi-rgb-led-matrix #298: Using ApplyStaticTransformer(UArrangementTransformer(1)) With Python

schreiberc, 2017-03-30:
I'm trying to run a 64x64 setup with 4 chained 32x32 panels.

I referenced this issue https://github.com/hzeller/rpi-rgb-led-matrix/issues/213
But it looks like edits have been made since then and it no longer applies.

I was able to get this working inside the example-api-use with the -L param. But there is no documentation in the python README.md file regarding ApplyStaticTransformer, and I don't know where or how to apply it inside my own custom python file. Or is this something that has to be set on the initial build? And if so where do i set it?

Thanks in advance for you help.

hzeller, 2017-03-30:
ApplyStaticTransformer has no equivalent in the Python library.

If you want to use the static transformer features in Python, you would have to manually add it in 
  https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/led-matrix.cc#L189
and
  https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/led-matrix.cc#L203
Then re-compile the library and re-compile and re-install the Python lib.

schreiberc, 2017-03-30:
Much appreciated!

closed 2017-03-30 by schreiberc

Arsenik4444, 2017-12-16:
Hi Sir,i wanted to ask how can i use or integrate the P10transformert in the python, this have to be add in the  /lib/led-matrix.cc#L189&led-matrix.cc#L203; I'm just confuse how this have to be added and done:i can run now demo Fine but still issues with the python,thank you