hzeller/rpi-rgb-led-matrix #299: Accessing  32x32Panel 1:8 scan rate

alexdisoza007, 2017-04-04:
Dear Henner,

I am facing the problem of accessing 32x32Panel , I gone through issue #66. I made change in transformer.cc and transformer.h  but I am not getting how make changes in demo-main text.cc.Please guide me how to make changes in these files. I written class like this:

`int LargeSquare64x64Transformer::TransformCanvas::width() const {
  return delegatee_->width() / 2;  }
int LargeSquare64x64Transformer::TransformCanvas::height() const {
  return delegatee_->height() * 2; }
void LargeSquare64x64Transformer::TransformCanvas::SetPixel(int x, int y, uint8_t red, uint8_t green, uint8_t blue) {
  int new_x, new_y;
  new_x = ((x / 16) * 32) + (x % 16);
  if ((y & 8) == 0) {
      new_x += 16; }
  new_y = ((y / 16) * 8) + (y % 8);
  delegatee_->SetPixel(new_x, new_y, red, green, blue);  }`

Thanks, 
Alex

possenhalter, 2017-04-29:
@alexdisoza007 I succeeded in getting [this](https://www.aliexpress.com/item/192x192mm-p6-led-panel-indoor-full-color-SMD-led-modules-32x32-pixel-1-16-scan-led/32537649016.html?spm=2114.13010608.0.0.vNArVX) 32x32 1:8 panel to work with the library.

I'm currently using the [Go binding](https://github.com/mcuadros/go-rpi-rgb-led-matrix), but I think this could be useful for you anyway.

I had to set the following parameters:

```go
rows       = flag.Int("led-rows", 32, "number of rows supported")
parallel   = flag.Int("led-parallel", 1, "number of daisy-chained panels")
chain      = flag.Int("led-chain", 2, "number of displays daisy-chained")
```
and use this transformation logic (I think it's same as yours):
```c
tx = x;
if ( y % 16 < 8 ) {
    tx = x + 32;
}
```
Hope it will help,
Zoltan

hzeller, 2018-01-21:
Please check out the new option `--led-multiplexing` https://github.com/hzeller/rpi-rgb-led-matrix#multiplexing
It does the mapping built into the library (and can also be used from Python.)
(You need to do a fresh checkout and compile, this just went in).

closed 2018-01-21 by hzeller