hzeller/rpi-rgb-led-matrix #300: Matrix + Photocell light sensor on GPIO 12

rginda, 2017-04-05:
I've got a photocell wired to GPIO 12 on an RPi3, as described in [this adafruit document](https://learn.adafruit.com/basic-resistor-sensor-reading-on-raspberry-pi/basic-photocell-reading) and have configured the Python rgbmatrix library as such:

```
  options.rows = 16
  options.chain_length = 3
  options.parallel = 1
  options.pwm_bits = 11
  options.gpio_slowdown = 3
  options.pwm_lsb_nanoseconds = 320
  options.drop_privileges = False
  options.hardware_mapping = "regular-pi1"
```

I've wired this all through your active adapter board.

I have a Python script that refreshes the matrix every 0.01 seconds.  If, during the refresh, I busy-loop wait for the GPIO to go LOW, as in the adafruit code, my readings vary wildly.   If instead, I refresh the matrix while waiting for the GPIO to go LOW, it goes LOW immediately. 

The standalone script in the adafruit doc works as expected and gives stable readings, and my matrix does not seem to be affected by my poking at GPIO 12.

I suspect the matrix library is changing the mode on GPIO 12, but the "regular-pi1" mapping doesn't specify a use for it, so I'm not sure why.

hzeller, 2017-04-05:
The matrix library is only configuring the GPIO pins to output that it needs for the operation, so if GPIO 12 is not configured for the matrix, you can read from it.

It looks like this works by setting the pin to output mode for a while and charging a capacitor, then reading back (and thereby changing the GPIO mode each time) this is an inherently noisy way to do it, also the waiting time there is probably very noisy as the Pi is busy. Doing this reading in particular if the pins around are switched with several megahertz to send to the matrix.

I'd look with a scope at it and see what is happening.

(why are you using regular-pi1, do  you still have such an old pi ? If you have a modern Pi, then if you set options.parallel = 1, then the other GPIO pins (e.g. 12) are not being used. If indeed you _do_ have a Pi1, then doing this reading while the Matrix is active might not work well because the CPU is borderline overwhelmed anyway)

rginda, 2017-04-05:
Thanks for the advise, I'll check with a scope.  Maybe it'll be better to skip this parts-bin hack for an actual light meter component.

I've got a pi3 with only a single chain attached.  I only went with "regular-pi1" because it doesn't use GPIO 12.

closed 2017-05-12 by hzeller