hzeller/rpi-rgb-led-matrix #301: 32x64 matrix output starts as expected, quickly becomes erratic

seb-rooks, 2017-04-06:
Thank you so much for all the work that you've put into this excellent library and it's excellent documentation. 

I'm sure that the problem I'm having is not a problem with your library, but after two days of trial and error, and searching, I can't find any video of anyone with output like mine.

It starts normal most of the time, then quickly becomes erratic in a few seconds. I've tried every example in the library, along with every combination of available flags, including brightness, pwm-bits, scanmode, slowdown, etc, none of which seem to have any consistent effect over approximately 200 trials. The timing of the output going erratic _seems_ completely random. But it's anywhere from the start, to about 10 seconds in.  After it goes wrong for a few seconds, it either goes completely blank, or stops on random patterns, holds for a few seconds, and then tries to continue on.

My refresh rate fluctuates between about 200 - 230hz

It may be worth noting that after it stops working as expected, it never returns to normal, unless I restart the demo, which then most of the time will work again as expected for up to about 10 seconds.

I'm powering with a computer psu. I'm not using a HAT, and I know my temporary breadboard setup is less than ideal, but it does work normally for nearly 10 seconds, which makes me suspect my problem lies elsewhere.

My matrix is 32x64 from ebay. The ic is SM245TS. I don't have a level shifter in place but I've tried reducing the voltage as suggested in the readme.

Could you please watch this short video and give me an idea of where my problem might lie? 

The first demo is suppose to say 'hzeller is awesome'. It's a shame it doesn't make it all the way through.

https://www.youtube.com/watch?v=gtyhXXXjxLE

Thanks

maddogs, 2017-04-06:
I had a similar problem to you.  In my case it I think it was the PIGPIO library I had installed http://abyz.co.uk/rpi/pigpio/

If you have a spare SD card, try installing a fresh OS, update etc and then then install the RGB Matrix library, disable sound etc and try again.

seb-rooks, 2017-04-06:
Many thanks @maddogs. I removed PIGPIO, rebooted, and reran the demos. Now, instead of working at all and becoming erratic, it goes straight to a random static configuration, and stays there.

I'm seeing that as a huge step in the right direction. At least something has changed. When I get home tonight i'll back my card up, wipe it, and install Raspbian lite (like the readme clearly states), and report back.

I should have tried that before posting, but I thought that my problem looked different than the problems of others that were solved by that, plus I like being able to launch idle with ssh -X

Thanks

seb-rooks, 2017-04-07:
Even though my problem appeared differently than the problems of others, running a fresh install of raspbian lite solved it. Let that be a lesson to anyone else trying to get away with running full raspbian.

Thanks for the help, and again for the awesome library. Now I'm off to figure out how I'm going to be able to incorporate this into a complex project without making it angry again.

https://www.youtube.com/watch?v=DbGFR8UT3Ek

closed 2017-04-07 by seb-rooks

hzeller, 2017-04-08:
Glad you got it working!