hzeller/rpi-rgb-led-matrix #302: HUB 12( example P10(1R) led matrix ) does not work

beikimajid, 2017-04-08:
hello,
I am getting a handful of 32x16 modules, chain them to a HUB12 and use as a display.
Has anyone fully accomplished such task in a way that you could share a "getting started" guide with wiring with only raspberrypi borad? I need sample code& wiring image.please help me.

hzeller, 2017-04-08:
The R and G correspond to the R and G on the Hub75. There is only one sub-panel, so you'd only connect R1 and G1.
A, B and C are the same.
One of the Latch or Output Enable are different polarity (as you mention), so you might need to reverse that (initially I would suggest just in hardware with an inverter (74HCT06), that way you don't have to fiddle with the code. I think the pulse subsystem can be made to output an inverted signal, but I'd start with a hardware inverter, which is more straightforward).

For the led switch, there is `--led-inverse`

Also, make sure to use a TTL signal converter (e.g. HCT245) - it might well be, that these panels can't deal with 3.3V signal levels.

I have not connected a Hub12 panel myself, but the inverted signal is something someone mentioned. So with that, you should see something.

Of course, you can go fancy and connect up to three times the number of Hub12 than Hub75 panels: just use R1 and G1 for the first panel, R2, G2 for the second panel and B1 and B2 for the third panel (and do manual color and pixel mapping in software). But that is for later :)

beikimajid, 2017-04-08:
Do you have an example of the picture of how to connect(P10(1R)) to Raspberry pin?
Can i connect led matrix(hub12) directly to raspberry without mediator?
Do you have a branches for P10(1R) to connect directly to raspberry pi with wiring & sample code?
I am very confused.
excuse me hzeller.

hzeller, 2017-04-08:
I do not have a Hub12 matrix, so there is no picture.

As I said above, the [wiring](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md) is the same as the Hub75, except that you leave out the B1, R2, G2, B2 and that the signals are on different pins. Ideally you use a second panel to connect to R2 and G2 and all the other pins in parallel witth the other panel, then you essentially simulate a Hub75 panel which internally has two sub-panels. 

And you need to invert the OE signal with an inverter (74HCT06, or simply with a transistor inverter). You can also change the software to generate a positive trigger, but it sounds like you are less comfortable doing that.

I don't know if you can connect your matrix directly to the 3.3V output, for this you have to look at the input chips on the Hub12 matrix if they can accept 3.3V TTL signal (see README about that).

It is very similar in general, once you wired it up, the example code will essentially work (except that there is no blue of course). This is pretty simple in general, you'll figure it out.

beikimajid, 2017-04-19:
Due to the My Post(https://github.com/hzeller/rpi-rgb-led-matrix/issues/303) & obtain the solution,
I use ./demo -D 0 --led-inverse without 74HCT06 FOR P10(1R)_HUB12 .
LED show things that I think I need to transform it.I use All transforms in issues But I did not answer.The general rule for display is correct But need to transform.
Please Help me

hzeller, 2017-04-19:
The help is in the [example code provided](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/transformer.cc). The transformer interface is really simple, so once you have figured out how the LEDs mapped for your particular panel, you can write it using the examples.

beikimajid, 2017-04-19:
hi hzeller,
[U-Arrangement Transformer] VS [RotateTransformer]  ?

closed 2017-06-04 by hzeller

NeerajShrivastava04, 2018-02-16:
hzeller,

Can you please explain that how to change the polarity of an OE signal in software...