hzeller/rpi-rgb-led-matrix #303: P5 smd show text but after six or more seconds became garbled

beikimajid, 2017-04-09:
Hello
I use P5 smd 32X32 1/16 scanning.All is ok just for few seconds but after six seconds or more became garbled(disturbed,damaged).Wiring IS OK.please Help Me

maddogs, 2017-04-09:
Have a read through this closed issue, it's similar to what you are describing

https://github.com/hzeller/rpi-rgb-led-matrix/issues/301

beikimajid, 2017-04-09:
Do I just have to running a fresh install of raspbian lite OR install PIGPIO on this Rasbian?
Thanks.

maddogs, 2017-04-09:
Try installing a fresh install of Rasbian Lite, update and upgrade. The install Henners Library and follow all the instructions including disabling sound.

beikimajid, 2017-04-10:
I try installing a fresh install of Rasbian Lite, update and upgrade and install Henners Library and disable sound But 32x32 matrix output starts as expected, quickly becomes erratic.Please Help Me.

hzeller, 2017-04-10:
* Make sure you don't have anything else running that uses GPIO (such as PiGPIO). Disable anything that could look like it also messes with GPIO.
 * Your power supply needs to be stable.
 * You didn't mention how you wired these up. Some matrix boards require that you do a TTL 3.3 -> 5V level conversion (see README)

beikimajid, 2017-04-10:
Hello hzeller
I don't have PIGPIO&Other that uses GPIO.
My Power cable : https://www.alibaba.com/product-detail/USB-TTL-3-3V-5V-Serial_60310219360.html

Wiring(follow all the instructions): 
R1	23	13	G1
B1	26	6	GRD
R2	24	21	G2
B2	19	6	GRD
A	15	16	B
C	18	22	D
CLK	11	7	LAT
OE	12	6	GRD

How to supply power to be stable?

Thanks a lot

hzeller, 2017-04-10:
Not sure what you mean with the power cable you mention, but a serial converter is of no use in this context.

You need TTL converter such as HCT245 chips, or build a board that is supplied in this project:
https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3

Did you read the troubleshooting section in the README ? https://github.com/hzeller/rpi-rgb-led-matrix#troubleshooting - there are some tips.

Also, if the matrix is too slow, try fiddeling with the slowdown `--led-slowdown-gpio`
If you do have some other thing that you don't know about but hogs some GPIO hardware pin, try `-led-no-hardware-pulse`

Regarding the power supply, there is a section in the readme that gives some thoughts.
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md#a-word-about-power

beikimajid, 2017-04-13:
hello hzeller.
My power supply is stable but i get same error.
I send the new film  >>> https://ufile.io/71d161

Please help me.

hzeller, 2017-04-13:
How did you measure to convince you that your power supply is stable ?
-> https://ufile.io/71d161 can you uplooad that to a more accessible site, such as YouTube ?

Some matrix work with 3.3V, but many need the conversion to 5V. So you are not doing level conversion now, you should try this now (also, please read the README that describes some experiments to try).

> Please help me.
I had several suggestions before, please try them first.

beikimajid, 2017-04-13:
Thank you hzeller for reply.
Which led matrix work with 3.3V?(for example)
Which non raspberry(nanopi,Orangepi, other ARM based) has 5v gpio pin like arduino?

hzeller, 2017-04-14:
It is hard to tell which matrix works with 3.3V - there are a miriad of manufacturers for it, so it really depends where you got it from. You really should make an attempt to read the README through, there are some tips how to determine that.

You will have a hard time finding a fast ARM based platform that uses 5V GPIO pins. 3.3V is already on the high side (besides, the other platforms would not be compatible with this library anyway).

In any case, you should build a TTL converter as provided in this project, so that you are not straining your computing device of having to drive long cables.

hzeller, 2017-04-14:
In your video it looks like you have some other process competing for GPIO.

beikimajid, 2017-04-15:
Hi hzeller.
I've found the solution!!!

sudo nano /boot/config.txt
disable dtoverlay=w1-gpio

Thank you for this Great Code

hzeller, 2017-04-15:
Ah, so some other thing was using the gpio. This is the first time I hear about it, maybe they added it in recent distributions ?
Thanks for the report back, I'll update the README.

beikimajid, 2017-04-15:
I think:

1.     3.3v raspberry pin
2.    Raspbian lite
3.    Pigpio
4.    Sound enable or disable

no affect on my project & this works well with passive pcb directly without 74hct245.

what is your idea?
Do you know what w1-gpio?

closed 2017-05-09 by hzeller