hzeller/rpi-rgb-led-matrix #304: Play Movie on LED Matrix

mmft1359, 2017-04-16:
Hi 
How can I play a movie on LED Screen?

hzeller, 2017-04-16:
If you ask questions, please make an effort to first read the README.

Did you look in the [client/](https://github.com/hzeller/flaschen-taschen/tree/master/client) directory ? There is a utility to play videos.

But I usually just use it with the FlaschenTaschen protocol (the FlaschenTaschen implementation has a switch to use this rgb-matrix), and then play movies via `vlc`, which has a FlaschenTaschen output mode (but I think you need to compile VLC from the latest source I think).

closed 2017-04-16 by hzeller

hzeller, 2017-05-28:
I have now included a video viewer that natively can play on the LED matrix:

https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/utils#video-viewer