hzeller/rpi-rgb-led-matrix #306: "text-example.cc" use cases to prevent flicker when piping output

chrisoutwright, 2017-04-19:
At the moment I have used the "text-example.cc" demo to display stock data with the ystockquote module via this command line (I have augmented the text-example internally to rotate and enable large display)  :

` (while :; do sudo python stockquote.py FDAXM7.EX FSMIM7.EX ; sleep 30; done) | sudo ./examples-api-use/text-example --led-rows 32 --led-chain 4  --led-slowdown-gpio=2 --led-pwm-lsb-nanoseconds 160 --led-pwm-bits=6 -y8 -f fonts/8x13B.bdf -C255,255,255`

The stockquote.py contains:

	#!/usr/bin/env python
	import ystockquote
	import sys
	import re

	def cleanhtml(raw_html):
	  cleanr = re.compile('<.*?>')
	  cleanr2 = re.compile('(\d+(\:)[\d]+)([a-zA-Z]+)([\s][-][\s])')
	  cleanr3 = re.compile('"')
	  cleantext = re.sub(cleanr, '', raw_html)
	  cleantext2 = re.sub(cleanr2, '', cleantext)
	  cleantext3 = re.sub(cleanr3, '', cleantext2)
	  return cleantext3

	def returnbid(arguments):
			i=0
			for tickerSymbol in arguments:

					i=i+1
					allInfo = ystockquote.get_all(tickerSymbol)

					if (i>=2):
					 print(tickerSymbol+":"+"\n"+str(int(float(cleanhtml(allInfo["last_trade_price"])))))

	if __name__ == '__main__':
		# test1.py executed as script
		# do something
			returnbid(sys.argv)

The only issue is that as soon stock quotes are gathered a mild flickering occurs. What could one do in setups like this to still enable fluid updates on the panels? Maybe seperate routine or something differently?

[image]

hzeller, 2017-04-19:
You should just write the output to the panel directly in C++ or Python. Calling it via the command line is enough overhead for the operating system that you can see the flicker. 

Since you anyway do the gathering of the data in Python, just add the few lines in Python to directly write it to the matrix (`graphics.DrawText(...)`), and do the sleeping and refresh directly in the Python program.

You need to install the [Python library](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/python) and might want to look in the [example directory](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/python/samples) for some examples how things are used.

chrisoutwright, 2017-04-19:
This sounds very reasonable to integrate the python code. I will use the sample `runtext.py`.
How exactly can I integrate in 'runtext.py':

    matrix->ApplyStaticTransformer(UArrangementTransformer(1));
    matrix->ApplyStaticTransformer(RotateTransformer(180));`

I did it in text-example.cc but not sure in python. Would you recommend integrating the stock quote gathering code in text-example.cc itself also, or would it be harder? Refreshing would be better in C++ right?

hzeller, 2017-04-19:
The Python code does not support setting transformers, sorry. You have to hack it manually, by adding
[here](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/led-matrix.cc#L189) and
[here](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/led-matrix.cc#L203).

If you feel comfortable, you can as well do the quote gathering in C++, but it is not strictly needed. The Matrix works reasonably well in Python, as speed critical Matrix code is executed in the C++ core.

chrisoutwright, 2017-04-19:
Thank you for the great help, I will try the modifications till next week, as a best friend's wedding is coming on and preperations are being called on me.
Concerning the transformer hack, this will enable U-form chaining (2 on 2 panels or 3 on 3)? 
Can rotation be worked into as well if manual rotating is limited because of cables on the wrong side?
Thank you.

hzeller, 2017-04-19:
Yes, you can call ApplyStaticTransformer multiple times. Whenever you change it, make sure to recompile and reinstall the Python library because it then is essentially "baked in".

hzeller, 2017-05-09:
Looks like you got it working ?

closed 2017-05-09 by hzeller