hzeller/rpi-rgb-led-matrix #307: "S" arrangement (ApplyStaticTransformer)

soyxan, 2017-04-23:
Hi,

I have the following physical connection:
`[<][<][<][<][<][<] }- Raspbery Pi connector`

But I want it to be treated as a 3 rows display:
```
[<][<] }- Raspbery Pi connector
[>][>]
[<][<]
```
I know that `ApplyStaticTransformer(UArrangementTransformer(...` will do the trick for an "U" arrangement, but what is the best way to achieve my "S" arrangement?

hzeller, 2017-04-23:
You essentially have to write a transformer yourself that does that. The interface is described in
   https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/canvas.h
.. and you get some inspirations in
   https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/transformer.h
.. and their implementation in
   https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/transformer.cc

(in general, if you have three rows, you really should consider actually connecting them in parallel. See https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3 for a helpful PCB you can build).

soyxan, 2017-04-26:
Thanks, I will try to implement my own transformer and let you know.

closed 2017-05-09 by hzeller

beikimajid, 2017-05-10:
Hi soyxan.
Do you have a result?

soyxan, 2017-05-10:
Yes, I implemented it in a way that you can specify how many "virtual rows" want to use. That is, for a "U" you pass 2, for a "S" you pass 3, and so on...

```
class SArrangementTransformer : public CanvasTransformer {
public:
  SArrangementTransformer(int parallel = 1, int virtualrows = 2);
  ~SArrangementTransformer();

  virtual Canvas *Transform(Canvas *output);

private:
  class TransformCanvas;

  TransformCanvas *const canvas_;
};

// S-Arrangement Transformer.
class SArrangementTransformer::TransformCanvas : public Canvas {
public:
  TransformCanvas(int parallel, int virtualrows) : parallel_(parallel), delegatee_(NULL), virtualrows_(virtualrows) {}

  void SetDelegatee(Canvas* delegatee);

  virtual void Clear();
  virtual void Fill(uint8_t red, uint8_t green, uint8_t blue);
  virtual int width() const { return width_; }
  virtual int height() const { return height_; }
  virtual void SetPixel(int x, int y, uint8_t red, uint8_t green, uint8_t blue);

private:
  const int parallel_;
  int width_;
  int height_;
  int panel_height_;
  Canvas *delegatee_;
  const int virtualrows_;
};

void SArrangementTransformer::TransformCanvas::SetDelegatee(Canvas* delegatee) {
  delegatee_ = delegatee;
  width_ = (delegatee->width() / (32*virtualrows_)) * 32;   // Div in middle at 32px boundary
  height_ = virtualrows_ * delegatee->height();
  if (delegatee->width() % (32*virtualrows_) != 0) {
    fprintf(stderr, "An S-arrangement would need an even number of panels "
            "unless you can fold one in the middle...\n");
  }
  if (delegatee->height() % parallel_ != 0) {
    fprintf(stderr, "For parallel=%d we would expect the height=%d to be "
            "divisible by %d ??\n", parallel_, delegatee->height(), parallel_);
    assert(false);
  }
  panel_height_ = delegatee->height() / parallel_;
}

void SArrangementTransformer::TransformCanvas::Clear() {
  delegatee_->Clear();
}

void SArrangementTransformer::TransformCanvas::Fill(
  uint8_t red, uint8_t green, uint8_t blue) {
  delegatee_->Fill(red, green, blue);
}

void SArrangementTransformer::TransformCanvas::SetPixel(
  int x, int y, uint8_t red, uint8_t green, uint8_t blue) {
  if (x < 0 || x >= width_ || y < 0 || y >= height_) return;
  const int slab_height = virtualrows_*panel_height_;   // one folded s-shape
  const int base_y = (y / slab_height) * panel_height_;
  y %= slab_height;
  if (y < panel_height_) {
    x += delegatee_->width() / virtualrows_;
  } else {
    x = width_ - x - 1;
    y = slab_height - y - 1;
  }
  delegatee_->SetPixel(x, base_y + y, red, green, blue);
}

SArrangementTransformer::SArrangementTransformer(int parallel, int virtualrows)
  : canvas_(new TransformCanvas(parallel, virtualrows)) {
  assert(parallel > 0);
}

SArrangementTransformer::~SArrangementTransformer() {
  delete canvas_;
}

Canvas *SArrangementTransformer::Transform(Canvas *output) {
  assert(output != NULL);

  canvas_->SetDelegatee(output);
  return canvas_;
}
```

beikimajid, 2017-05-11:
Hi soyxan.
thanks a lot for reply.
Is your code works for Mix Mode for **1-Parallel**? as example :

Two "S" arrangement & 1-Parallel
[<][<] }--- Pi connector
[>][>]
[<][<]-- S
[>][>]
[<][<]
[>][>]--S

one "S" & one "U" arrangemenet & 1-Parallel
[<][<] }--- Pi connector
[>][>]
[<][<]--S
[>][>]
[<][<]--U

yvesgarnier, 2017-05-17:
Dear Soyxan
I have implemented your code on my pi(The first part in transformer.h and the second part in transformer.cc)

I have a 3x3 Panel setup chains as an S  (3 virtual rows with a chain length of 9 panels). The image is now showing but kind of messed up. Do you may know how to solve this problem?

Thanks a lot!

chrisoutwright, 2017-06-12:
Dear Soyxan,
as yvesgarnier has pointed already out, the transformation is not quite working out right. I tried it myself.
Is it working in your setup? How did you wire the panels?

Thanks a lot.