hzeller/rpi-rgb-led-matrix #309: Missing transformer?

Czajo, 2017-04-23:
Hi again! After my fight with hub12 I bought 32x16 rgb led matrix with hub75 i think.
I have now:
R1 || G1
B1 || no description
R2 || G2
B2 || no description
A || B
no description || no description
CLK || STB
OE || LR_OE

I tried regular connection, but I have effect like on this films:
Demo #0 https://youtu.be/8zpTB6BmYFo
Demo #1 https://youtu.be/iZRUuyk5zFY
What I doing wrong?

hzeller, 2017-04-23:
It looks like one of these outdoor panels that have internal zig-zag mapping. They use 1:4 multiplexing, so choose rows=8 and chain=2 - internally, it pretends to be a thing with 64 pixels long, but only 8 high. You can see at the 'backwards running', that they are esentially folded around. You need to write a transformer for this (there are some entries in the issue tracker where people already did that, but it is probably faster to do it from scratch than scouring through there).

The second issue is in your video is, that things seem to 'stop' in-between. Make sure that there is no other program or daemon that attempts to access the GPIOs. Recently someone discovered that there was an issue with the 1-wire protocol, that apparently is enabled in some instances: switch that off.

Make sure that there is no pigpio daemon running.

In general make sure to install the simplest operating system, I suggest Raspian Lite.

LR_OE ? I don't know as well :)

hzeller, 2017-05-09:
Did you get it to work ?

closed 2017-07-30 by hzeller