hzeller/rpi-rgb-led-matrix #310: Flickering and slow image

villabeto, 2017-04-24:
Dear your work in this library is invaluable. Thanks a lot of. 
Sir, I'm running your library in raspberry pi b+, and I developed a code in Python to show my own pictures. But, I have some problems: 
- The picture charge very slow
- The displays are blinking (turn off and turn on) each second approximately. 

You can see my display in this link: https://youtu.be/xls6Z1qfo8Q

The code is:
```
import time
import sys
from scipy import misc
from rgbmatrix import RGBMatrix, RGBMatrixOptions
from PIL import Image
import numpy
import os.path
if len(sys.argv) < 3:
    sys.exit("enter two valind image links")
else:
	image_file1 = sys.argv[1]
	image_file2 = sys.argv[2]
	if not os.path.isfile(image_file1):
		 sys.exit("enter two valind image links")
	if not os.path.isfile(image_file2):
	 	 sys.exit("enter two valind image links")

img = Image.open(sys.argv[1]).convert('RGB')
img.save('im1.png')

img = Image.open(sys.argv[2]).convert('RGB')
img.save('im2.png')

imarray1 = misc.imread('im1.png')
imarray1=misc.imresize(imarray1,(64,192,3))
ima1=misc.imrotate(imarray1[0:31,:,:],180)
ima2=imarray1[32:63,:,:]
ix1=numpy.concatenate((ima1,ima2), axis=1)
imarray2 = misc.imread('im2.png')
imarray2=misc.imresize(imarray2,(64,192,3))
ima1=misc.imrotate(imarray2[0:31,:,:],180)
ima2=imarray2[32:63,:,:]
ix2=numpy.concatenate((ima1,ima2), axis=1)
ima=numpy.concatenate((ix1,ix2), axis=1)
im = Image.fromarray(ima.astype('uint8')).convert('RGB')

options = RGBMatrixOptions()
options.rows = 32
options.chain_length = 24
options.show_refresh_rate =1
options.parallel = 1
options.pwm_lsb_nanoseconds=270
options.hardware_mapping = 'regular' 
options.pwm_bits=5
options.scan_mode=0
options.brightness=80
matrix = RGBMatrix(options = options)
matrix.Clear()
a=1
matrix.SetImage(im)
time.sleep(300)

```

There are two displays as the follow picture. Each display have 6 RGB modules (32x64 px).
**How can I solve it?** **Thanks!!**
Regards, Alberto.

[image]

hzeller, 2017-04-24:
## Observations 
 * 24 panels on a single string is a lot, and a lot of data has to be pumped for each update. 
 * The Raspberry Pi B+ is fairly slow (It has the same CPU as the Raspberry Pi 1). The changing the picture speed is probably due to that.
 * Python is slow in _particular_ on the Pi 1 (which is the same as the B+ model you have. https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/python#performance ). It is 10x slower than on a Pi 3.
 * Just skimming at the code, you seem to do the pixel rotating and concatenating where to put pieces of the image in Python, to account for the cabeling of the panels. This is very slow. A [transformer]( https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use#remapping-coordinates) on the C++ side can do that essentially for free.
 * If you are using 100% CPU straight, the kernel will interrupt the process regularly as I have observed in other cases - this is the regular blanking you see. Using 24 Panels in one chain will use 100% CPU on a Pi 1.

## What you should do

  * Get a recent Raspberry Pi 3 - it has a much faster CPU than the Pi1 and has 4 cores. So not only is it faster to update the panel, the matrix update can happen in parallel to your image preprocessing. Using a Pi3 will have a huge positive impact.
  * I suggest not to put 24 panels (or 12 double 32x64 as it seems for you) in one long chain. Rather, use the feature to drive parallel chains of panels with shorter chains. That way, you need three chains of 8. The good thing is, that driving the panels in parallel does not use more CPU than having a single chain of 8. So that is strongly recommended in this case. You can make it simple by building this adapter: https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3
  (then use the setting `options.parallel = 3` and `options.chain_length = 8`). 
 * The Pixel-mapping is better done with a transformer: a class that tells the library [where each pixel is to find](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use#remapping-coordinates). You have to write one or adapt an existing transformer do match your mapping. These are unfortunately in C++, so you have to do that there, then plug this transformer manually into [here](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/led-matrix.cc#L189) and [here](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/led-matrix.cc#L203), and recompile the Python library. That way all the expensive rotating/concatenating you had done in Python is not something needed anymore: you just consider this as one large canvas of 192x192 pixel where each pixel is out put where it should be.
  * If you can choose between Python 2.7 and Python3, choose Python 2.7 - it is a little faster.
  * (If you can, you might consider switching to C++ as it will provide more speed. But more important are the other points above).

Bottom line: getting a Pi 3, doing the parallel panel connecting and adding a transformer will make this a wonderful, much fast image viewer.

villabeto, 2017-04-25:
Thanks a lot of again Mr. **hzeller**. I'll get a Raspberry Pi 3 and test the displays on it. 
¿But now, how can I call your library using C++?, I can't see it..
I was testing a lot of and now I found a solution with this parameters. 

```
options = RGBMatrixOptions()
options.rows = 32
options.chain_length = 24
options.show_refresh_rate =1
options.parallel = 1
options.pwm_lsb_nanoseconds=280
options.hardware_mapping = 'regular' 
options.pwm_bits=6
options.scan_mode=0
options.brightness=80
matrix = RGBMatrix(options = options)
matrix.Clear()
a=1
matrix.SetImage(im)
```
Now, the flickering is very very fast. I need to see in detail the screen for see that effect. 
In other case, thanks for your suggestions, I'll follow your instructions. 

Regards, Alberto.

hzeller, 2017-05-09:
The library is primarily C++, so you can use that directly (see `include/` directory).
Python is only a wrapper for the C++ library.

closed 2017-05-09 by hzeller