hzeller/rpi-rgb-led-matrix #311: 64x32 1/8 scan panel

gpulido, 2017-04-24:
Hello,
After the test with a p10 32x16 panel 1/4 scan I have bought some p5 64x32 1/8 scan panels (this [https://www.aliexpress.com/item/Outdoor-Waterproof-IP65-Pixel-Pitch-5mm-Ultra-Bright-320mm-x-160mm-Full-Color-1-8scan-LED/32796979637.html])
And now I'm trying to figure out which (if any) transformation matrix should I apply to it.
However the first step is to try to iterate througth all the pixels at least once even not in the "right" order.
I've tried with row size of 32 and num chains 2 and any time I light one coordinate two positions are iluminated on the matrix, and that is the best I can do.
It seems that the software is not being mapping corretly this configuration.
Also a lot of flickering is shown on the matrix.
Is anything that I can do to make it work properly?

Thank you in advance
Regards

beikimajid, 2017-04-26:
Hi
How to use p10 32x16 panel 1/4? What is your transformer?Please tell me

gpulido, 2017-04-27:
Hello @beikimajid you can found it here: https://github.com/hzeller/rpi-rgb-led-matrix/issues/283#issuecomment-279859048

Also you can review the entire issue if you want to learn how I get to it.

gpulido, 2017-04-30:
Hello,
After playing a little with the panel I have found a configuration that at least would paint all pixels:
I have to set the parameters num or rows to 16 and the chained panels to 4.
This provides me to a 128x16 matrix that I can at least traverse. Also I have to add the --led-slowdown-gpio 2 to avoid flickering and some strange behavior.

This is the result:

[https://goo.gl/photos/79H3fjaTCreqqybd9](url)

Now I have to figure out the transformation matrix for this panel.
I will keep you informed

gpulido, 2017-05-01:
Hello,
After some tunning I have found the transformer to make this work.
@hzeller If you want me to test some code onto my panels so you can integrate them in the library please let me know.
The following transformer works for P5 64x32 1/8 outdoor (at least with the pixel array configuration that I have)
```c
void P5outdoorTransformer::TransformCanvas::SetPixel(int x, int y, uint8_t red, uint8_t green, uint8_t blue) {
  int odd8_block =(((y % 16) / 8) +1) % 2; //to know if the pixel is in an odd 8pixel row
  int new_x = (x+ 64 * odd8_block); 
  int new_y = (y % 8) + 8 *( y / 16);
  delegatee_->SetPixel(new_x, new_y, red, green, blue);
}
```

beikimajid, 2017-05-02:
Hello gpulido,
As I said before, in [https://github.com/hzeller/rpi-rgb-led-matrix/issues/316](url),this code supports the boards have this property:
Scan_Multiplexing = (2 / ROWS_of_panel)

Do your transform code support 64x32 1/8 compeletly?
Can you attach a picture of your work that show ./demo -D 0 --led-rows=16 --led-chain=4?
Do your transform code support 32x16 1/4 also?

gpulido, 2017-05-02:
hello, this is an output video:
[https://goo.gl/photos/8L6UW2FZdWPoHj1m7](url)
This transformer only works for the P5 64x32 panel that I own. The configuration of the panel is shown in the first video of this issue, where I just paint each pixel in order without transformations.

Regarding the P10, the transformation matrix that is on the other issue is the one that works for that panel. 
You have to figure out which kind of panel you have, by iterating and after that try to found the right trasnformation matrix.

Hope it helps.

gpulido, 2017-05-02:
My previous transformer was wrong as it didn't take into account the daisy chained panels.
This is the right one:

```c
void P5outdoorTransformer::TransformCanvas::SetPixel(int x, int y, uint8_t red, uint8_t green, uint8_t blue) {
  int odd8_block =(((y % 16) / 8) +1) % 2; //to know if the pixel is in an odd 8pixel row
  int num_mod_x = (x / 64); //num of daisy chainned modules
  int new_x = (x + 64*num_mod_x + 64*odd8_block); 
  int new_y = (y % 8) + 8 *( y / 16);
  delegatee_->SetPixel(new_x, new_y, red, green, blue);
}
```
Sorry for the mistake

beikimajid, 2017-05-03:
I have a [P10 16x32 1/4 scan].
I looked #https://github.com/hzeller/rpi-rgb-led-matrix/issues/283#issuecomment-279859048 and run [P10outdoorTransformer] with [./demo] But my panel show disordered then I use this Transformer:

int new_x = 0;
int new_y = 0;
if (y < 4 || (y > 7 && y < 12)) {
new_x = (x * -1) + (24 * ((x / 8) + 1)) - 17;
new_y = y;
} else {
new_x = x + (8 * ((x / 8) + 1));
new_y = y - 4;
}

this work with [./demo runtext16.ppm --led-chain=2 --led-rows=16] BUT [./demo -D 0] appears wrong.

Can you tell Where is my mistake?
Can you attach **mapping** of P10 16x32 1/4 scan that You mentioned above?
Can you attach a picture of your work that show result on P10 16x32 1/4 scan ?
 
My Mapping It seams : 
[P10.xlsx](https://github.com/hzeller/rpi-rgb-led-matrix/files/972471/P10.xlsx)

yigitbayol, 2017-11-15:
hello @gpulido @hzeller,
first of all thanks for sharing P5 outdoor transformer. 
I try to use your problem, but I have problem.

**Here what i did:**
1- I added this code to transformer.h

> class P5outdoorTransformer : public CanvasTransformer {
> public:
>   P5outdoorTransformer(int angle = 0);
>   virtual ~P5outdoorTransformer();
> 
>   void SetAngle(int angle);
>   inline int angle() { return angle_; }
> 
>   virtual Canvas *Transform(Canvas *output);
> 
> private:
>   // Transformer canvas to rotate the input canvas in 90° steps
>   class TransformCanvas;
> 
>   int angle_;
>   TransformCanvas *const canvas_;
> };

2- Also i added this code to transformer.cc

> class P5outdoorTransformer::TransformCanvas : public Canvas {
> public:
>   TransformCanvas(int angle);
> 
>   void SetDelegatee(Canvas* delegatee);
>   void SetAngle(int angle);
> 
>   virtual int width() const;
>   virtual int height() const;
>   virtual void SetPixel(int x, int y, uint8_t red, uint8_t green, uint8_t blue);
>   virtual void Clear();
>   virtual void Fill(uint8_t red, uint8_t green, uint8_t blue);
> 
> private:
>   Canvas *delegatee_;
>   int angle_;
> };
> 
> P5outdoorTransformer::TransformCanvas::TransformCanvas(int angle)
>   : delegatee_(NULL) {
>   SetAngle(angle);
> }
> 
> void P5outdoorTransformer::TransformCanvas::SetDelegatee(Canvas* delegatee) {
>   delegatee_ = delegatee;
> }
> 
> int P5outdoorTransformer::TransformCanvas::width() const {
>   return delegatee_->width() / 2;
> }
> 
> int P5outdoorTransformer::TransformCanvas::height() const {
>   return delegatee_->height() * 2;
> }
> 
> void P5outdoorTransformer::TransformCanvas::SetPixel(int x, int y, uint8_t red, uint8_t green, uint8_t blue) {
>   int odd8_block =(((y % 16) / 8) +1) % 2; //to know if the pixel is in an odd 8pixel row
>   int num_mod_x = (x / 64); //num of daisy chainned modules
>   int new_x = (x + 64*num_mod_x + 64*odd8_block); 
>   int new_y = (y % 8) + 8 *( y / 16);
>   delegatee_->SetPixel(new_x, new_y, red, green, blue);
> }

3- Finnaly i activate ApplyStaticTransformer from led-matrix.cc : 
> ApplyStaticTransformer(P5outdoorTransformer());

After these, i rebuild python and C codes : 

>  make -C lib clean all
   sudo make -C python clean
   make build-python
   sudo make install-python

when i try to run image-scroller.py, there was an error like this:

>  File "led_manager.py", line 4, in <module>
    from samplebase import SampleBase
  File "/home/eczane/core/bindings/python/samples/samplebase.py", line 7, in <module>
    from rgbmatrix import RGBMatrix, RGBMatrixOptions
  File "/usr/local/lib/python2.7/dist-packages/rgbmatrix/__init__.py", line 7, in <module>
    from .core import RGBMatrix, FrameCanvas, RGBMatrixOptions
ImportError: /usr/local/lib/python2.7/dist-packages/rgbmatrix/core.so: undefined symbol: _ZN10rgb_matrix20P5outdoorTransformerC1Ei

how can I solve this? I don't understand why.. 
If i disable ApplyStaticTransformer from led-matrix.cc, there is no error but also there is no transform.
Also i don't see any error while compiling.

Thanks.

dma19, 2017-11-20:
Hello, Merhaba @yigitbayol,
P5outdoorTransformer(int angle = 0); must be P5outdoorTransformer(); 
and you will delete some line about the rotation, we don't need this.

RobertKrikke, 2018-01-15:
I tested P5outdoorTransformer with:
examples-api-use/demo -D0 --led-rows=16 --led-chain=4 --led-parallel=2

Works fine, thanks @yigitbayol and @dma19

GOBish, 2018-01-20:
Hi RobertKrikke,
Thanks for posting your reply. I also have an outdoor LED that I am having trouble with. It is a Linsn P5, 1:8. Details are here: 
http://www.ledcontrolcard.com/outdoor-led-module-c-49/latest-outdoor-p5mm-18scan-64x32dots-320mmx160mm-smd-led-module-p-578.html
I haven't been able to get the code you added to fix the issue.
I am using an adafruit hat, so I can't use parallel = 2.
Where in the files you mentioned did you add your code?

Any and all help is appreciated.

RobertKrikke, 2018-01-20:
> latest-outdoor-p5mm-18scan-64x32dots-320mmx160mm-smd-led-module-p-578.html

Hi,

I pushed my code to:
https://github.com/RobertKrikke/rpi-rgb-led-matrix

Remember that other led-matrixes are not working anymore...

Robert

angelogoncalve, 2018-01-20:
Use for P5outdoorTransformer with:
examples-api-use/demo -D0 --led-rows=32 --led-chain=2 --led-parallel=1

GOBish, 2018-01-20:
thanks for the reply, but that solution didn't work.
The board is a P5 outdoor module, 32x64 made up of two 32x32 panels ribbon-ed together. I think it is 1:8 scan. They seem to be set up differently, though, in that each 32x32 has some kind of routing in it.
I tried the -L option (for U shape) for both the whole 32x64 board and for just one of the 32x32 panels, but that didn't help.

angelogoncalve, 2018-01-20:
Try use one panel 32x32 1:8 scan rate. The transformer for your case is:
void P5outdoorTransformer::TransformCanvas::SetPixel(int x, int y, uint8_t red, uint8_t green, uint8_t blue) {
int odd8_block =(((y % 16) / 8) +1) % 2; //to know if the pixel is in an odd 8pixel row
int num_mod_x = (x / 32); //num of daisy chainned modules
int new_x = (x + 32*(num_mod_x + odd8_block)); 
int new_y = (y % 8) + 8 *( y / 16);
delegatee_->SetPixel(new_x, new_y, red, green, blue);
}

Use for one P5outdoorTransformer with:
examples-api-use/demo -D0 --led-rows=16 --led-chain=2 --led-parallel=1

RobertKrikke, 2018-01-20:
I just tested the software with one panel (64x32) and it still works:
https://nl.aliexpress.com/item/P5-outdoor-waterdichte-volledige-kleur-led-display-64x32-pixel-320x160mm-panel-1-8-scan-smd-2727/32836168595.html?spm=a2g0s.9042311.0.0.EZkMRt

I'm not an expert, but I think if you try to fix it, try to get only one panel working. The second panel should also work without any changes otherwise I don't think you can chain them.

GOBish, 2018-01-20:
Thanks very much for the reply - I'll try it out on my end and let you know the results

hzeller, 2018-01-20:
BTW, per popular demand, i bought some 1:8 panels and just added two typical transformers found in 1:8 panels. You can choose them with `--led-multiplex=1` (for some strip mapping)  or `--led-multiplex=2` (checker-pattern thing).
Just do a fresh git pull, compile and try these options (on c++, not passed through to Python yet).

GOBish, 2018-01-20:
That is great news HZ! Thanks for doing that. I will try it out for my setup, I hope it works.

RobertKrikke, 2018-01-21:
It works on one panel:
examples-api-use/demo -D0 --led-rows=32 --led-cols=64 --led-multiplexing=1 --led-parallel=1
and more panels:
examples-api-use/demo -D0 --led-rows=32 --led-cols=64 --led-multiplexing=1 --led-parallel=1

Great work!

@hzeller --led-multiplex is not working, according your code it should be multiplexing or the other way around. In the usage of the programs also --led-multiplex is used.

hzeller, 2018-01-21:
Ah, thanks for noticing the typo. Will fix it.

GOBish, 2018-01-21:
Got it working! Thank you all so much for the follow up!

To get my panel working as 32x64, I had to use the following:
--led-rows=32 --led-cols=32 --led-chain=2 --led-multiplexing=2 --led-parallel=1

I'm getting a bunch of ghosting, but now I've got a place to start.

GOBish, 2018-01-21:
Again, thanks for the help.

I know, asking for more, but - any advice on getting this to work with the python samples?

hzeller, 2018-01-21:
Ghosting is often a function of the gpio slowdown or pwm nanoseconds; see the Readme for tips in that regard.
Wiring up through Python I'll add later tonight.

hzeller, 2018-01-21:
The flags are now available in Python. Have fun!

closed 2018-01-21 by hzeller

GOBish, 2018-01-21:
in runtext.py I get this error:

File "runtext.py", line 35, in <module>
    if (not run_text.process()):
  File "/home/pi/OutdoorLED/rpi-rgb-led-matrix/bindings/python/samples/samplebase.py", line 44, in process
    options.cols = self.args.led_cols
AttributeError: 'rgbmatrix.core.RGBMatrixOptions' object has no attribute 'cols'

hzeller, 2018-01-21:
You have to recompile and re-install the python binding.

GOBish, 2018-01-21:
OK, thought I did but will do it again. Thanks again for all your help

hzeller, 2018-01-21:
Make sure to call a `sudo make clean` in the python directory before recompiling. Maybe something was sitting around.

GOBish, 2018-01-21:
hmm, still getting the same error even after recompiling

hzeller, 2018-01-21:
and re-installing with `sudo make install-python` ?

GOBish, 2018-01-21:
still no go

hzeller, 2018-01-21:
So I just tried a fresh checkout on some Pi and it compiles and runs cleanly with the Python2.7 there. So I guess there is something in your set-up.
Try a fresh checkout.

GOBish, 2018-01-21:
ok will do, thanks for checking into this

hzeller, 2018-01-21:
Yay, wonderful. Have fun with your now functioning LED matrix :)

kshitij283, 2024-04-19:
We are also facing the current problem, can you share the library which you used for mega