hzeller/rpi-rgb-led-matrix #313: 1/4 scanning not work

beikimajid, 2017-04-26:
Hello
All documentation in this Repository not work for 16x32 (1:4 multiplaxing) even https://github.com/gausie/rpi-rgb-led-matrix/tree/p10transformer.I test All but I did not answer
Please Help me to write transform Conceptually.
thank you

beikimajid, 2017-04-27:
I find Solution.this work for me and show runtext16.ppm with chian 2 , rows=16 , scan 1/4:

int new_x = 0;
    int new_y = 0;
    if (y < 4) {
        new_x = (x * -1) + (24 * ((x / 8) + 1)) - 17;
        new_y = y;
    } else if (y > 7 && y < 12) {
        new_x = (x * -1) + (24 * ((x / 8) + 1)) - 17;
        new_y = y;
    } else if (y > 3 && y < 8){
        new_x = x + (8 * ((x / 8) + 1));
        new_y = y - 4;        
    }else if (y >=12){
        new_x = x + (8 * ((x / 8) + 1));
        new_y = y - 4;        
    }
    delegatee_->SetPixel(new_x, new_y, red, green, blue);

It work on one rows perfectly. How to show just one runtext16.ppm(large) on 3 panel that wire By column.Do I have to write transform?How?

P1>P2>P3  (NO)

(YES)
P1
|
P2
|
P3

gpulido, 2017-05-01:
hello @beikimajid 
It seems that you panel is the same as the one I configured on the issue:
[https://github.com/hzeller/rpi-rgb-led-matrix/issues/283](url)
Take a look to that issue on which everything is explained

hzeller, 2018-01-21:
Please check out the new option `--led-multiplexing` https://github.com/hzeller/rpi-rgb-led-matrix#multiplexing
It does the mapping built into the library (and can also be used from Python.)
(You need to do a fresh checkout and compile, this just went in).

closed 2018-01-21 by hzeller