hzeller/rpi-rgb-led-matrix #314: Estimated Performance Hit for Per-Panel Transformation

Timelord0, 2017-04-27:
My supplier changed up their panels.  The new panels have green/blue channels swapped.  Is the processing cost of a per-panel channel flipper going to be significant, or shall I rest easy?

Given the four new panels are ever so slightly brighter/blueish, I will probably be adding some per-panel calibration code as well if needed.

The project is running Hub75e 1/32 64x64 panels; 2 of one batch, 4 of the second batch.  While I suppose it would be possible to simply buy a couple replacement panels from new batch, I would prefer a software solution that can be used in future to correct any issues.

I have not really played with transformers yet, so I am not entirely sure the built in capabilities of your library.  I look forward to the challenge, however, and thank you again for this wonderful library!

hzeller, 2017-04-27:
* green/blue channel swap:
   Overhead is probably negligable. 
   You would need to hack that manually in the code, as a Transformer does not look at color swap modifications - it only looks at coordinates (see point at end).
  What you have to do, is to manually swap the green/blue depending on the coordinates you get in the SetPixel() call: roughly here https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/framebuffer.cc#L309 - then recompile the library with that patch.

 * There, you can possibly also multiply the values you get with some constant to adapt brightness.

 * Usually, it is a good idea to have the same batch of panels for avoiding-headache-reasons, but it is of course possible to hack around it with software. However, as described above, you have to manually hack this into the library (at the aforementioned position), so it is somewhat on the ugly side.

 * You can't use Transformers unfortunately - they only work with remapping coordinates not colors. They once were just implemented as delegation, which could've allowed that, but color changing was almost never something done, so the current code eliminates the overhead (everything is calculated at startup-time)

Timelord0, 2017-04-27:
Alright.  I'll see how it goes, thanks!

closed 2017-04-27 by Timelord0