hzeller/rpi-rgb-led-matrix #317: Lower half of the LED Matrix Board doesn't show green colors

ritvikbiswas, 2017-05-02:
Hello,

Everything works fine for me except the colors on the bottom half of the display. For some reason, the bottom half's LEDs won't display the color green. I verified this by running the first demo and then noticing that half of the green "Hello" text was cut off. 

How can I solve this? 

Thank You,
Ritvik Biswas

hzeller, 2017-05-02:
Do you run it on some first generation Pi? Did the remedy described in the Readme not help?

ritvikbiswas, 2017-05-02:
I ran it on the pi 3 model b. Which g2 line should I try to correct?

hzeller, 2017-05-02:
The one in the chain that is not working. If you only have one chain, that would be Pin 21 to the panel.

https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md#wiring-diagram

ritvikbiswas, 2017-05-03:
Thank you for your help. A simple change of wires at Pin 21 solved my problem.

closed 2017-05-03 by ritvikbiswas