hzeller/rpi-rgb-led-matrix #318: 16x32 Matrix

lllllIll, 2017-05-05:
I Have a single 16X32 matrix connected to Raspi 2, using the wiring diagram you detail.

I run "sudo ./rotating-block-generator.py" and this is the result... what could possibly be wrong here? I did the wiring twice and got the same result. Pi is not overclocked or anything.

Result : 
[https://drive.google.com/file/d/0Bxv7_b-cG0beR1NDeDdEdlB0elU/view?usp=sharing](url)
 
Some picture of panel and wire: https://drive.google.com/open?id=0Bxv7_b-cG0beTmZocjBqTEdTenM

hzeller, 2017-05-06:
Make sure to read https://github.com/hzeller/rpi-rgb-led-matrix#troubleshooting
the strange stopping of the image looks like you have the issue with the one-wire configuration.
Also make sure to install a minimial Raspbian Lite.

hzeller, 2017-05-06:
(also for your display you would give it the parameter `--led-rows=16` as it is only 16 high)

lllllIll, 2017-05-06:
I just make checkup the troubleshooting section and 
[image] I think it's Logic level voltage not sufficient It's possible ?

hzeller, 2017-05-06:
Logic level and driving long lines from the RPi logic outputs are always a concern, and I encourage to build a level converter like this https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3

In your case however it looks like logic levels are actually ok as the panel is showing things. So as first step I suggest to make sure that

   * the `--led-rows=16` is set
   * make sure that nothing else on the Raspberry Pi is attempting to use the GPIO pins (e.g. PiGPIO or 1-wire configuration).

lllllIll, 2017-05-06:
THANK YOU I reinstall raspbian lite on raspberry pi and now it's work !!!!

closed 2017-05-06 by hzeller