hzeller/rpi-rgb-led-matrix #319: 64*64 RGB LED Matrix

Anatoli113, 2017-05-08:
Hi Hzeller,
I was really impressed with data in your git repository.
It really helps people who want to control the LED matrix.
I am currently working on a 64 * 64 matrix.
To do this, I designed and tested an adapter board (active) based on your data.
I used 220V/5V AC/DC adapter for LED Matrix.
Adapter board is supplied from the RPI2.

The path of the test code is as follows. 
Https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/python

As a result of this code, the image was initially displayed correctly. However, the screen wanes sharply even after 3 to 5 seconds.
I also run your demo-main.cc, but the result is the same.
I used your sample as it is.
I think there is something I can not think of.

I want you to help me solve quickly this problem.
I will wait for your good reply.
I'm so sorry to have taken your busy time.
But I believe that you will help me.

Best regards!!!
anatoli.

hzeller, 2017-05-08:
Did you read the Troubleshooting section ? Image going away typically means that you have something that also fights for the GPIO pins, so make sure to

  * run a Raspbian Lite
  * not have configured anything like PiGPIO or one-wire configuration

Anatoli113, 2017-05-08:
Thanks for your effort.
I solved this problem.
I have not removed dtoverlay=w1-gpio in /boot/config.txt
The one-wire protocol was enabled (w1-gpio).
So I disabled it. 
Then the LED works correctly.

Thanks for your help.

Best.

closed 2017-05-09 by hzeller