hzeller/rpi-rgb-led-matrix #320: Display video with Utilities

casaservice, 2017-05-09:
hello
please i need help, in the prject you said "The utils directory is meant for ready utilities to show images, animated gifs, text and video. Read the README there for instructions how to compile." but i don't find a documentation about how tu display video
please i need help !

aycgit, 2017-05-17:
hzeller just changed it in commit bd095ac. You can display videos using flaschen-tachen but the pi's build in video decoder isn't fast enough is what hzeller stated. I personally dont know how much work the pi does with flaschen-taschen but at this moment there is no command line tool for videos.

hzeller, 2017-05-17:
Here is an example how to do video locally -
 https://gist.github.com/hzeller/524310b05065bd3e1aa41a8e26574815
But it is too flickery with the software decoder. If someone can get ffmpeg to use the hardware decoder of the Pi, that might be a different story.

aycgit, 2017-05-17:
oh cool, didnt know that existed, i was trying to do it using opencv but it was still too slow.

hzeller, 2017-05-28:
I have now added my basic video viewer to the distribution. I tested that the flicker is actually not a problem when the video is transcoded first to the right size; so I documented that.

Also, it supports a new feature that allows to store the raw, preprocessed stream of frames to disk and then replay it later. This allows for very fast frame rate even with larger displays without extra use of decoding CPU at play-time (at the expense of disk, that is).

Check out
https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/utils#video-viewer

closed 2017-05-28 by hzeller