hzeller/rpi-rgb-led-matrix #321: 128x128 led matrix with U-arrangement & one parallel

beikimajid, 2017-05-10:
Hello.
I have 8 led matrix 32x32 (total 128x128) & 1 raspberry with one raspberry(1-parallel)  that Join with U-arrangement:   
  Pi connector #1}--- [>][>] 
                               [<][<]
                               [>][>]
                               [<][<]

Which transformer code should I use to support [./demo -D 0] to show square fully?

hzeller, 2017-05-10:
You would need to write one, it is pretty straightforward, have a look at the examples in [transformer.cc](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/transformer.cc). You just need to think of what physical coordinate should be mapped to the original underlying coordinate.

In general, I'd recommend doing this with two parallel chains however, because you degrade refresh rate for such a long arrangement (because with the 64 long panels, it is essentially 16 panels long).

closed 2017-05-10 by hzeller

beikimajid, 2017-05-11:
Hi hzeller.
Can you tell me more with example for "physical coordinate & original
underlying coordinate".
as example if we have two 32x64 panel Sort by rows, how to mapped them.
(0,0) > ???
(0,1) > ???

I still have not learned basic.
Thanks a lot.

chrisoutwright, 2017-06-13:
[I have done it for 96x96](https://github.com/hzeller/rpi-rgb-led-matrix/issues/341) if it is any help.