hzeller/rpi-rgb-led-matrix #322: How to display video via 64*64 Led matrix?

Anatoli113, 2017-05-11:
Hi Hzeller,

I want to display short video via 64*64 Led matrix.
There are many video extensions.(for example avi, mov)

I would like to know where the reference data for this is in your repositary.
Maybe I want to know how to do it quickly.
I hope your good reply.

Best.
Anatoli

hzeller, 2017-05-12:
I have built a video-viewer, but the ffmpeg decoding library uses CPU a lot so stealing some needed to do the display update. So the resulting picture is very flickery. The Pi is not fast enough to do the decoding and do a smooth play.
I therefore decided not to include the code as it is not a good example. You can [try yourself](https://gist.github.com/hzeller/524310b05065bd3e1aa41a8e26574815), but I won't give any support.

These days to view videos with the RGB matrix, I usually just run the library with the [FlaschenTaschen protocol](https://github.com/hzeller/flaschen-taschen/tree/master/server#rgb-matrix-panel-display) and then send video from another computer using vlc (which supports FlaschenTaschen protocol if you compile it from head).

closed 2017-05-12 by hzeller

andygrace, 2017-05-12:
Interesting and decent first stab at an FFmpeg integration, Herr Zeller! 

I might take a closer look at this as some of my r&d dev is based around getting ffmpeg to work with HWAccel on mobile clients, particularly iOS and Android. 
Latest versions of FFmpeg > 3.1  support the Broadcom/MMAL abstraction layer for H.264 decode.   Audio would still be an issue unless you piped it out to a codec chip or something.

I still think your FlaschenTaschen solution is probably better if you have a decent PC in the vicinity.

hzeller, 2017-05-12:
With hardware decoding, things might be decent, but is that available/accessible in the Pi SoC?

andygrace, 2017-05-12:
Yes, MMAL is the abstraction layer for the Broadcom H.264 hardware decoder 

For more details of HW abstractions for FFmpeg all platforms...
https://trac.ffmpeg.org/wiki/HWAccelIntro

And the wrapper for RPi in userland is here 
https://github.com/raspberrypi/userland/tree/master/interface/mmal

Cheers, Andy

hzeller, 2017-05-12:
If you can make a decent video decoding experience on the Pi so that it can play on the RGB Matrix ... I accept pull requests @andygrace :)

hzeller, 2017-05-28:
I have now included the simple video viewer in the distrubution:
   https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/utils#video-viewer

In case there is too much flicker, it seems to be sufficient to first transcode the video into the right size to reduce CPU use at decoding time. Also, there is now LED matrix specific streaming format that stores raw streams that can be used as well. So given that there are solutions to the original problems that let me be hesitant to include this in the distribution, I've finally submitted it.

pchehade, 2018-04-19:
Hello @hzeller ! 

I was able to use photoshop to create a 32x32 pixel size video in a .mp4 format. When I tried to use video-viewer I received the following error:

[swscaler @ 0xd2e940] No accelerated colorspace conversion found from yuv420p to rgb24. 

For my use case, I plan to store very short video files locally on the pi to play on the matrix - the vlc option is not ideal for me. What can I do differently?

greigs, 2019-11-25:
@pchehade did you ever manage to get that yuv420p format working? @hzeller does video-viewer only support input files with RGB24 format?