hzeller/rpi-rgb-led-matrix #325: make install-python errors

Lorrainbow, 2017-05-20:
Hello,

I am trying to get my LED matrix running with Python. The sample code for C works fine. 
I have followed the steps and got as far as:

sudo make install-python PYTHON=$(which python3)

What exactly do I need to enter?
I've tried "sudo make install-python PYTHON=python3" and I get the following errors:

make -C ./lib
make[1]: Entering directory '/home/pi/rpi-rgb-led-matrix/lib'
make[1]: Leaving directory '/home/pi/rpi-rgb-led-matrix/lib'
make -C python build
make[1]: Entering directory '/home/pi/rpi-rgb-led-matrix/python'
make -C ../lib
make[2]: Entering directory '/home/pi/rpi-rgb-led-matrix/lib'
make[2]: Leaving directory '/home/pi/rpi-rgb-led-matrix/lib'
3 setup.py build --build-lib .
make[1]: 3: Command not found
Makefile:67: recipe for target 'build-python' failed
make[1]: *** [build-python] Error 127
make[1]: Leaving directory '/home/pi/rpi-rgb-led-matrix/python'
Makefile:33: recipe for target 'build-python' failed
make: *** [build-python] Error 2

Thank you for your help

Lorrainbow, 2017-05-20:
me again!

I installed PyPy
I ran the "sudo make install-python PYTHON=python3" - and got no errors

I tried to run the sample "sudo ./runtext.py --led-chain=1" 
which didn't work

I replaced it with
"sudo python3 runtext.py --led-chain=1" which did work

hzeller, 2017-05-21:
Yep, the reason is because runtext.py runs just 'python', which in your system is python2.7. Since you compiled with Python3, you need to also run Python3.

Your first make failed, because it tried to execute a binary '3' (`make[1]: 3: Command not found`), so I guess you had some issue when you exeucted the `which python3`.

In general, Python3 is still a bit slower than Python2.7, so you might consider using that unless you really need Python3.

closed 2017-05-21 by hzeller