hzeller/rpi-rgb-led-matrix #327: I want to configure 320x256 LED matrix.

Anatoli113, 2017-05-22:
Hi Hzeller,

I am going to build a large Led matrix chain using p2.5mm 64x64 led matrix or p5mm 32x32 led matrix.
Its size is 320x256.
Therefore, when I am going to use 64x64 led panels, I think I will have to configure 4 parallels.
Each parallel contains 5 panels.
I want to know my thought is right.
And I know it can configure only 3 parallels using RPI2 and RPI3.
If so, how to configure for 320x256 size?

If there is another way to configure it, how to define the device in my code to display an image or video in that configuration?

Already I have made an active3 adapter board.
I hope you will help me.
I will wait for your good reply.

Best regards.

Anatoli juny.

hzeller, 2017-05-24:
320x256 is a pretty high resolution, you probably will only get 80Hz refresh rate.

You can only do three parallels, so you would need to map the 20 panels into three parallels, e.g. by making three parallels with 7 panels each (but one is not connected, so instead of 21 panels they are 20). Since each 64 panel is internally behaving as two long, you need to do `--led-chain=14`.

Now you need to do a particular layout in which you arrange these panels and write a pixel [mapping transformer](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use#remapping-coordinates) which you need to write specifically for your use-case.

But again, this resolution is borderline too high for a reasonable refresh rate.

Anatoli113, 2017-05-24:
Hi, 
Thanks for your help.
But to be honest, I still do not know how to configure a 5x4 (320x256) with a 64x64 Led panel.
I drew a block and send it for understanding. It is what I am thinking.
 
  [<][<][<][<][<]  }--- Pi connector #1                            
  [>][>][>][>][>]
  [<][<][<][<][<]  }--- Pi connector #2
  [<][<][<][<][<]  }--- Pi connector #3

[<][<][<][<][<]  }--- Pi connector #1                            
[>][>][>][>][>]
[<][<][<][<][<]  }--- Pi connector #2                          
[>][>][>][>][>]

Which configuration is right?
If both aren't right, how to configure the 320x256?

Currently, I tested to display image on 64x64 led module with python. I used your image viewer code, it is working correctly.
I am going to do programming with python for the testing of 320x320 LED matrix.
I already saw https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use 
I know the canvas transformer is c library.

I do not know how to refer it to a 320x256 LED matrix when implementing Python code.
(I only used your code when display the images with python.)

I would really appreciate it if you could give me specific instructions on how to do it.

Best regards.

Anatoli.

hzeller, 2017-05-24:
Configure 3 subchains with 7 each. You would need to start a chain in the middle of the display, something like

```
[<][<][<][<][<]  <1)
[>][>] (2>[>][>][>]
[<]<3) [<][<][<][<]
[>][>][>][>][>]
```

With the pixel-mapping, you can do anything so you can map that back to behave like a square display.

Once you have written the transformer in C++, it is not yet available in Python, correct. For now, you have to manually add it in the c++ led matrix constructors, and then recompile your Python library to be able to use it there ([here](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/led-matrix.cc#L184) and [here](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/led-matrix.cc#L198) ).

Given that Python is pretty slow and you have to update a lot of pixels, I strongly suggest to use C++ in your later project.

Anatoli113, 2017-05-25:
Thanks for your help.

Anatoli113, 2017-05-25:
I want to know how much power is needed for 20 of the 64x64 matrix panel.
I know it is needed 3.5A per 32x32 led matrix.
If it is needed about 3.5A per 64x64 matrix, it will be needed about 3.5*20=70A for 20 of the 64x64 panel.
Is this right?
I hope that you will give me the correct solution for the supplying power.
And please tell me the capacity value and voltage value of polar capacitor to connect to the power terminal of each 64x64 panel.

Best.

Aantoli.

hzeller, 2017-05-25:
A 64x64 panel has 4x as many LEDs than a 32x32 panel, so it might use more; however the LEDs are also smaller so that reduces that estimate again. So it is hard to estimate, you have to measure the current consumption of your specific panels. This depends on the vendor etc., so this is what you have to do first: set one of the 64x64 panels to all white (write a little program, that uses `Canvas::Fill(255, 255, 255)`) and measure the current it is using.

The exact description and calculation how to calculate the thickness of the wires and capacitors is then given here:
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md#a-word-about-power

Voltage rating of the capacitor should be anything about 6.3V, typically that would be 16V or 25V. Try to get Low ESR capacitors.

Anatoli113, 2017-05-29:
Hi Hzeller,
I get a lot from your advice. Thanks for you.
I am going to use the BPI-M64 Banana Pi to control our led matrix.

https://www.aliexpress.com/item/64-bit-Quad-core-mini-single-board-computer-BPI-M64-Banana-Pi-Board/32766257704.html?spm=2114.01010208.3.7.zw4S7Z&ws_ab_test=searchweb0_0,searchweb201602_5_10152_10065_10151_5010011_10130_10068_10136_10157_10137_10060_10138_10155_10062_10156_437_10154_10056_10055_10054_10059_303_100031_10099_5190011_10103_10102_10096_5170012_10147_10052_10053_10107_10050_10142_10051_10084_10083_10080_10082_10081_10178_10110_519_10111_10112_10113_10114_10037_10182_10185_10032_10078_10079_10077_10073_10123_142,searchweb201603_1,ppcSwitch_3&btsid=71f5e555-5c04-4378-a351-923352b7c237&algo_expid=661d895e-0277-41bf-a5af-85a3f9ba96b3-0&algo_pvid=661d895e-0277-41bf-a5af-85a3f9ba96b3

I see it is better than raspberry pi.
The interface of pinout is same the raspberry pi.
So I am going to use the adapter board.

I want to know if I need to modify current code and configuration used on the raspberry pi when I use the BPI-M64.

I will wait for your reply.
Best.

anatoli.

hzeller, 2017-05-29:
I am not sure if the Banana Pi will work, so far I know they have a completely different GPIO subsystem, so it might likely not work. I have never played with a Banana Pi, so I don't know.
You need to read the datasheet of the processor used in the Banana Pi to understand how their GPIOs are working. Unless you commonly read datasheets and internal hardware descriptions and write code for it, it is probably not worthwhile.

The Raspberry Pi3 is plenty good. In fact it is twice as fast as they LED matrix can handle typically, so there is not much you'd get with another computer that claims to be faster.

closed 2017-06-04 by hzeller

Anatoli113, 2017-06-12:
Hi Hzelller.

Anatoli113, 2017-06-12:
I configured 256x192 led matrix using 64*64 led module and displayed the video on it.

hzeller, 2017-06-12:
You absolutely **do not** want to use Python to fill such a large display. Python is way too slow for this.

This is something you need to do in C++ and use double buffering (writing to an off-screen canvas and switching it with the `SyncOnVSync()` call). You might even need to implement it with triple-buffering to avoid wasting any CPU cycles while waiting for the next frame.

Even then, you are limited by the memory bandwidth the processor allows for setting pixels (which is a complex operation), which is around 4Megapixels/sec on a Raspberry Pi 3, so the fastest you can set pixels and update pictures is about 80 frames/sec (which is good enough to display a 60p video, but probably does not leave much margin to create the pixels in the first place).
(You can go much faster by preparing frames and store/restore them preprocessed using the [content stream utility classes](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/content-streamer.h). )

For video viewing, you might just consider using the [provided `video-viewer` utility](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/utils#video-viewer). In its documentation, there are also hints of how to get the best performance, e.g. already have the video input encoded in the right size. The video-viewer also allows to write a content stream file (discussed above), which then can be read by the `led-image-viewer`.

Anatoli113, 2017-06-12:
Hi Hzeller,

hzeller, 2017-06-12:
The video-viewer is doing exactly this, so just have a look there.