hzeller/rpi-rgb-led-matrix #328: Can't solve my ghosting issue

NAM1025, 2017-05-23:
First, thanks for the awesome library. I've been using it for a while and it's great. Love the new updates!

That being said...

I'm using the AdaFruit Hat(with hardware hack to reduce flickering ) and a 16x32 board. I've been using an older version of the library for a while, but noticed some ghosting. While looking it up I saw that you had updated everything and figured I'd do the same (Changing parameters without recompiling is nice). The ghosting issue actually got worse with the newest version and even added  a pulsating effect to any LEDs where ghosting was present. I was able to reduce it to what was on the old version with the commands suggested in the 'Read Me', but even with nanoseconds around 1500 (any higher and the refresh rate would be visible to the human eye) it was still present.

I did notice that the ghosting occurs only when the color green is present... so any teal, yellow, white, or green will show ghosting. It doesn't happen at all with red, blue or any combination of the two. 

Before I go and buy a new LED board or buy a new Matrix Hat/have one of the custom ones made, is there anything else I can try?

hzeller, 2017-05-23:
mmh, 1500ns is pretty high.
Interesting with the ghosting being worse with green present, it sounds like the current control they have on green is sensitive - this is the first time I have heard, but I only have seen ghosting rarely, it seems to be an issue with certain panels. Then again, our eyes are most sensitive to green, so a low value of green ghosting is more visible.

What I would try is to make sure that there is good power to the panels - good thick wires and possibly one or two low ESR capacitors directly at the board. I don't know exactly where the ghosting is coming from, but I suspect it is some ringing in the internal current source, possibly exaggerated from suboptimal power; so it might make sense to test if things change with improved power.

NAM1025, 2017-05-23:
Thanks for the quick reply and sorry forgot to mention that I have it set to 300ns now, but at 1500ns ghosting was still present. 

For reference: Here's a (crappy) video of it at its worst before I started messing with settings.
https://gfycat.com/LameFlawedDuiker

I'll take a look at the board in the morning and see if I can swap out the wires powering it to see if it helps.

closed 2017-06-04 by hzeller

NAM1025, 2017-06-09:
Hey, sorry I'm reopening the thread, but I haven't had much free time. 

I finally took everything apart over the weekend and tried some new wires for powering the LED board, but I'm still see the ghosting. I ended up buying a new LED board, hooked it up yesterday, and have the same issue. Is there anything else I can try?

hzeller, 2017-06-09:
Did you buy a different LED board, or from the same manufacturer ?

NAM1025, 2017-06-09:
Bought the same board from Adafruit. In retrospect I guess I should have tried a different company. 

Current setup with the least amount of ghosting I could find (while retaining a decent refresh rate)..
pwm_bits = 8
led-pwm-lsb-nanoseconds=650
gpio_slowdown=1

Refresh rate is ~75Hz

chrisoutwright, 2017-06-14:
I would try: 
`--led-slowdown-gpio=2 --led-pwm-lsb-nanoseconds 130 --led-pwm-bits=11`

Do you also have the _pwm hack_ for adafruit HAT or do you use _regular gpio_ mapping?