hzeller/rpi-rgb-led-matrix #329: rgb led panel flickers to  random colours

arvindpandi6, 2017-05-23:
sir,
we have built an rgb panel similar to adafruit 32x32. we have used  tlc5927 as driver,we have connected 1 chainlink and 2 in parallel ,i have attached the image that is displayed on screen 
[image]
i have attached the open cv code i use to read images 
#include <opencv2/core/core.hpp>
#include <opencv2/imgproc/imgproc.hpp>
#include <opencv2/highgui/highgui.hpp>
using namespace cv;

int main(int argc, char *argv[])
{
      while(1)
    {
    Mat image;
    image=imread("skull.jpg");
    if(image.empty())
      return -1;

    imshow("lena", image);
    if(waitKey(1)>0)break;

}
}

the panel is flickering i couldnt  figure it out whether it is due to refresh rate or clock speed ,can you pls help me

arvindpandi6, 2017-05-23:
I ALSO USED IT TO GIVE LIVE VIDEO USING THE FOLLWOING CODE THE SAME PROBLEM PERSISTS EXISTS

#include "opencv2/opencv.hpp"
#include "led-matrix-c.h"
#include <stdlib.h>
#include <stdint.h>
#include <stdio.h>
#include <string.h>
#include <unistd.h>
//#include <pthread.h>
using namespace cv;

void rotate(Mat& src, double angle, Mat& dst)
{
Point2f pt(src.cols/2., src.rows/2.);
Mat r = getRotationMatrix2D(pt, angle, 1.0);
warpAffine(src, dst, r, cv::Size(src.cols, src.rows));
}
int main(int, char**)
{
///////////////////////////////////////////////////////////
 struct RGBLedMatrixOptions options;
  struct RGBLedMatrix *matrix;
  struct LedCanvas *offscreen_canvas;
  int width, height;
  int x, y, i;

  memset(&options, 0, sizeof(options));
  options.rows = 32;
  options.chain_length = 2;
  options.parallel=1;
matrix = led_matrix_create_from_options(&options, NULL, NULL);
  if (matrix == NULL)
    return 1;
 offscreen_canvas = led_matrix_create_offscreen_canvas(matrix);

  led_canvas_get_size(offscreen_canvas, &width, &height);

  fprintf(stderr, "Size: %dx%d. Hardware gpio mapping: %s\n",
          width, height, options.hardware_mapping);
///////////////////////////////////////////////////////////

   VideoCapture cap(0); // open the default camera
    if(!cap.isOpened())  // check if we succeeded
        return -1;

    Mat edges;

    for(;;)
    {
        Mat finale;
        cap >> finale; // get a new frame from camera
        imshow("edges",finale);
        if(waitKey(1) >= 0) break;
        resize(finale,finale,Size(32,64));
       // imshow("resized",finale);

      int rows = finale.rows;
int cols = finale.cols;
int largest = 0;
if ( rows > cols ){
largest = rows;
}else{
largest = cols;
}
Mat temp = Mat::zeros(largest, largest, CV_8UC3);

// Copy your original image
// First define the roi in the large image --> draw this on a paper to make it clear
// There are two possible cases
Rect roi;
if (finale.rows > finale.cols){
roi = Rect((temp.cols - finale.cols)/2, 0, finale.cols, finale.rows);
}
if (finale.cols > finale.rows){
roi = Rect(0, (temp.rows - finale.rows)/2, finale.cols, finale.rows);
}

// Copy the original to the black large temp image
finale.copyTo(temp(roi));

// Rotate the image
Mat rotated = temp.clone();
rotate(temp, 90, rotated);

Mat result = rotated(Rect(roi.y, roi.x, roi.height, roi.width)).clone();

//imshow("rotated", result);

       int z=0;
        for(int y = 0; y <64; ++y)
          {
          //Option 1: get a pointer to a 3 -channel  element
          cv::Vec3b* pointerToRgbPixel = result.ptr<cv::Vec3b>(y);
               for (int x = 0; x < 64; ++x, ++pointerToRgbPixel)
               {

                    uint8_t blue =  (*pointerToRgbPixel )[0];
                    uint8_t green = (*pointerToRgbPixel )[1];
                    uint8_t red =   (*pointerToRgbPixel )[2];

                     //final_array[z+1]=red;
                     // final_array[z+1]=green;
                    //final_array[z+2]=blue;
                    z=z+3;

               led_canvas_set_pixel(offscreen_canvas, x, y, red, green, blue);
               }
          }
         offscreen_canvas = led_matrix_swap_on_vsync(matrix, offscreen_canvas);
    }
    led_matrix_delete(matrix);

    // the camera will be deinitialized automatically in VideoCapture destructor
    return 0;
}

hzeller, 2017-05-23:
So you re-build how these 32x32 panels work, with all the clocking and latching, dual row adressing and clock-in, essentially recreating the HUB75 bus ? Cool.

You should start using the demo examples in the distribution to see if things work properly there. With `--led-show-refresh` you see the refresh rate created by the library. For a 32x64 it should be in the 200Hz range, so not really flicker.

If the demos are working, then something is going on with your software; I found for instance that decoding videos on the Pi is too slow and it uses up CPU that make things flicker.
If the demos are already flickering, then you need to look back what your panel is doing. Since you build it, you should be able to trace what is happening on there with an oscilloscope.

hzeller, 2017-05-23:
Also make sure to provide enough capacitors near each of your TLC5927 for it to be able to regulate properly.

arvindpandi6, 2017-05-24:
sir ,
as you mentioned it is 200 hz range for 32 *64 panel,we have the same ghosting problem which is shown in this video with the reference to the ghosting thread here 

https://gfycat.com/LameFlawedDuiker

we have also used 0.1microfarad  for tlc driver and 220 microfarad for power supply to filter..

i am not sure how to change the refresh rate in this code,will changing then refresh rate solve this problem??

hzeller, 2017-05-24:
You can slow down the output with the `--led-slowdown-gpio` flag.

220uF sounds pretty low to me. Since you built this, you can calculate how much storage capacity you need for the overall current drawn.

I don't know if refresh rate will solve the problem - if you have built the matrix, you know the exact electrical characteristics so you should be able to spec the right values.

arvindpandi6, 2017-05-24:
i tried slow down flag also but i couldnt come to an concluision ,will changing the spi clock work,is there any posibility to change spi clock ,

hzeller, 2017-05-24:
There is no SPI being used.
The clock speed is determined by the led-slowdown-gpio.

Did you measure with your oscilloscope ?

hzeller, 2017-05-25:
Is it always flickering or only after a couple of seconds ? In that case, read the Troubleshooting section in particular the part about the one-wire setting.

arvindpandi6, 2017-05-25:
sir,
i tried gpio slowdown but i couldnt find any i am pro adapter am directly gib=vingvement,i am not using adafruit hat hub75 adapter for these panels as they using hc245 level shifter ..is hat necessary??

arvindpandi6, 2017-05-25:
sir,
i tried gpio slowdown but i couldnt find any improvement , this panel is built according to standard adafruit panel with hc245 level shifting..i am not using adafruit hat hub75 adapter for my panels as i am using hc245 level shifter ..is hat adapter necessary??

hzeller, 2017-05-26:
You need a HCT245 to be compatible with 3.3V voltage level.

arvindpandi6, 2017-05-30:
sir,
i have solved the ghosting issue ,,,but i find the panel blinks in,,,is there any way to stop the blinking

arvindpandi6, 2017-05-30:
i mean once the gpio has slowed down i can see a slower refresh rate,how to solve it,

hzeller, 2017-05-30:
You are not saying how fast it 'blinks' or what are the settings you gave. Or how many panels you have connected, what is your power supply. You need to give enough information to be able to assess what your issue is.

So I assume you followed all the parts in the Troubleshooting section (using minimal Raspbian, only logged in remotely to Pi with no local UI, using the isolcpus settings for the kernel, have a solid power supply with short supply lines, measured with an oscilloscope that there are now dips in your power etc....).

If you have to slow it down so far that you see flickering and only use a single panel, then you should get a better panel. These panels should work with high switching rates and if there is ghosting that means that their internal current controller is badly reacting even if the power is good.

arvindpandi6, 2017-05-31:
yup,i checked and got succedded the panel now works fine ,i have connected two panels in parallel ,the panels are working fine now,the error was in hardware only we have recitifed those and its working fine

thanks for your guidance and immediate support which helped me to accomplish it

hzeller, 2017-05-31:
Cool!

closed 2017-05-31 by hzeller