hzeller/rpi-rgb-led-matrix #331: Cannot seem to find the correct setting for my panel.

garydion, 2017-05-28:
Hello all,

I'm pulling my hair out trying to find the correct settings for my new panel here, so I'm turning to you all for help.  First, here is a link to the panel I purchased:
[https://www.arrow.com/en/products/mikroe-2347/mikroelektronika](url)
And here is the supposed datasheet from Digikey for the unit (though the picture doesn't match exactly):
[https://media.digikey.com/pdf/Data%20Sheets/MikroElektronika%20PDFs/P6-8S_Indoor_LED_Display_Tech_Parameters.pdf](url)

I'm using a Pi3 and the Adafruit Hat.  And this is what I'm seeing...

With the following line:
sudo ./demo -D 0 --led-no-hardware-pulse --led-gpio-mapping=adafruit-hat

I see: 
[https://youtu.be/Ulgj8dsz1bg](url) 
...which seems to indicate that the first two quarters are ghosted on top of one another, and the last two do the same.  This made me think my screen had a hardware issue.

But then, I played around with some settings (which I know are totally wrong) with the following line showing a little promise:
sudo ./demo -D 0 --led-no-hardware-pulse --led-gpio-mapping=adafruit-hat --led-rows=64 --led-chain=2
And it dispelled my concern over hardware because the ghosting is gone:
[https://youtu.be/5U6hqIlP6lE](url)
The flickering is in the camera, btw; it's smooth to my eyes.

So, is it obvious to anyone what I'm doing wrong?  I appreciate any help!  

Thanks, Gary

hzeller, 2017-05-28:
From the pictures, it looks like you got a panel with 1:8 multiplexing. Essentially the LED panel internally maps the pixels a bit differently. We need to unmap that to have a nice square picture again.

First, you need a transformer that can deal with it; I don't have any yet in this distribution (I hope to get around doing that soon). You can try one provided by other users of this library, e.g. https://github.com/hzeller/rpi-rgb-led-matrix/issues/283#issuecomment-279859048

Then, as flags, you need `--led-chain=2 --led-rows=16`

garydion, 2017-05-28:
Excellent feedback!  Okay, I can 100% confirm that `--led-chain=2 --led-rows=16` works with this panel because the following travelling-dot program shows all 1024 lights are individually addressable (though the rows are out of order):
`  for (int disp_y = 0; disp_y < 16; disp_y += 1)
  {
    for (int disp_x = 0; disp_x < 64; disp_x += 1)
    {
      canvas->SetPixel(disp_x, disp_y, 200, 200, 200);
      usleep (10000);
      canvas->SetPixel(disp_x, disp_y, 0, 0, 0);
    }
  }
`
I will try gpulido's transformer and report back as to whether it worked or needs to be modified. 
Thanks again!

hzeller, 2017-05-28:
You might need to fiddle with it a little, I think gpulido's transformer was for a half-height 16 pixel panel, while you have a 32 high. So you might need to double some numbers in strategic places :)

garydion, 2017-05-28:
Okay, so it sees the display as 16 lines high and 32 columns wide.  So, as I move the pixel with the above code, it races from left to right across Line 9, then 1, then 10, then 2, and so on until Line 8 is complete.  Then, it jumps to line 25, then 17, then 26, then 18, and so on.  Since I have the pattern down now, it shouldn't be too hard to transform.  I'll be back shortly...

garydion, 2017-05-28:
I must apologize for not quite understanding what your code is expecting... do I need to write my transformer to convert from a standard 32x32 to the 64x16 my display needs or from the weird 64x16 pattern to a normal 32x32?

hzeller, 2017-05-28:
The code presents itself to be a Canvas of 32x32, which then calls the underlying canvas that has size 64x16.

garydion, 2017-05-28:
Okay, it will take me a while to figure out how to properly integrate it into your project's transform files, but the following snippet of code works great for converting a single-module canvas of 32x32 to the underlying 64x16 canvas the flavor of display I have wants:
```
   int new_x = x;
   int new_y = y;
   if (y < 8) 
      new_x = x + 32;
   else if (y < 16) 
      new_y = y - 8;
   else if (y < 24) 
   {
      new_x = x + 32;
      new_y = y - 8;
   }
   else 
      new_y = y - 16;
```

garydion, 2017-05-29:
Happy news!  I made the necessary updates to transformer.cc and .h and added the following to demo-main.cc on line 1163 after the check for rotation:

```matrix->ApplyStaticTransformer(rgb_matrix::Mikro2347Transformer());```

Now, the following command results in the desired rotating square:

```sudo ./demo -D 0 --led-no-hardware-pulse --led-gpio-mapping=adafruit-hat --led-rows=16 --led-chain=2```

Thank you so much for your help!  AND for code that I'm startled I was able to modify with your and other's help.  I've never written C++ before.  :)

hzeller, 2017-05-29:
Wonderful, congratulations!

There is always a first time programming in a new programming language, and I am glad you made it through with a result that works.

(If you add it before the rotation as the first transformer, then rotation will also work, as it then just works as lowest level in the chain.).

For now, if you want to use the led-image-viewer in the utils/ directory, you also have to add your code snippet adding the static transformer there. I plan to make these things a little simpler, but until then, this is necessary :)

closed 2017-05-29 by hzeller