hzeller/rpi-rgb-led-matrix #333: Active 3 board 5v/gnd pads suggested connector

gpulido, 2017-06-01:
Hello,
I would like to use the 5v /gnd pads of the Active 3 board to feed the rpi, however, there is not a suggested connector to be used with the pads.
I have several kf301 but they are very big, so what is a good connector type that I could use?
Thank you in advance
Gabriel

hzeller, 2017-06-01:
There is no suggested connector, just solder on the cables.

hzeller, 2017-06-01:
FYI, the distance is 3.81mm, which is one of the common formats for connectors, a bit smaller than 5mm that the KF301 has. You might find something with this distance on eBay or something.

gpulido, 2017-06-01:
@hzeller Thank you very much, that is exactly the info that I need.
It would be nice to add it to the active-3 wiki page so it is accesible.

closed 2017-06-01 by gpulido