hzeller/rpi-rgb-led-matrix #335: Every other pair of LEDs Unlit

abarlam, 2017-06-03:
I have an issue where every other pair of lines aren't lighting up, e.g. lines 1+2 are lit, 3+4 aren't, 5+6 lit, 7-8 unlit, etc. I checked my solder joints, and even used a multimeter to check continuity and everything is good. I have a 64x32 panel.

Video - https://www.youtube.com/watch?v=1tCMy3edgIw

I thought maybe I had a bad ribbon cable, so I wired the pins myself and also tried an other ribbon cable. Same issue.  I also have another one of these panels with a similar model number P5 pitch (aliexpress same vendor) and it experiences the same problem.  I am using adafruit rgb hat, and have also done the flicker mod.  Same issue with and without the mod.  This is on a pi3.

hzeller, 2017-06-04:
It sounds like your `B` line is not properly connected, but that would mean that pixels 3+4 would be folded into 1+2 which it doesn't look like.
Maybe your board has a slightly different pin-out that does something different with the odd line-pairs ? Do you have a picture of the connector on the back of the board in which it is possible to read the labels on that connector ?

FWSquatch, 2017-07-26:
Just wanted to let you know that I had the exact same issue as abarlam. You were correct. It did turn out to be the B line that was connected wrong. Thanks for the help!

closed 2017-07-30 by hzeller